ros-industrial/ros_canopen #53: Driver dies on start-up

thiagodefreitas, 2015-01-16:
After starting the driver with the following command:
```roslaunch cob_bringup canopen_402.launch can_module:=can3 component_name:=arm_left```

core service [/rosout] found
process[arm_left/arm_left_driver-1]: started with pid [11549]
terminate called after throwing an instance of 'boost::interprocess::interprocess_exception'
  what():  Permission denied
process[arm_left/arm_left_joint_state_controller_spawner-2]: started with pid [11564]
[arm_left/arm_left_driver-1] process has died [pid 11549, exit code -6, cmd /u/robot/git/care-o-bot/devel/lib/canopen_motor_node/canopen_motor_node joint_states:=/joint_states __name:=arm_left_driver __log:=/u/robot/.ros/log/94872932-9d5f-11e4-a05f-eca86bff61c1/arm_left-arm_left_driver-1.log].
log file: /u/robot/.ros/log/94872932-9d5f-11e4-a05f-eca86bff61c1/arm_left-arm_left_driver-1*.log
process[arm_left/control_mode_adapter-3]: started with pid [11565]
[ INFO] [1421402705.666247316]: waitForService: Service [/arm_left/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1421402710.684724028]: waitForService: Service [/arm_left/controller_manager/load_controller] has not been advertised, waiting...
^C[arm_left/control_mode_adapter-3] killing on exit
[arm_left/arm_left_joint_state_controller_spawner-2] killing on exit
[WARN] [WallTime: 1421402712.897394] Controller Spawner couldn't find the expected controller_manager ROS interface.
[arm_left/control_mode_adapter-3] escalating to SIGTERM
shutting down processing monitor...

Possible reasons ?
Shared memories are not released properly?
SocketCAN interface not functional

ipa-mdl, 2015-01-16:
This is a problem with shared memory (used for multi-chain support).

What happened before?
Did the driver die? Did you switch the user running the driver?

thiagodefreitas, 2015-01-16:
It dies on start-up, without even initializing.

The steps for reproducing the error are not yet know.

Once it happens, the error can only be fixed with rebooting t1,

ipa-fmw, 2015-01-16:
@ipa-mdl: you're right. this happened after a previous crash of the driver with another user account. Is there some way to prevent this?
- either allow all users to overwrite
- or shutdown the node correctly if it crashes

ipa-mdl, 2015-01-16:
Let's do both :)

ipa-fmw, 2015-01-22:
done by @ipa-mdl

closed 2015-01-22 by ipa-fmw