ros-industrial/ros_canopen #54: Failure during switch routine

thiagodefreitas, 2015-01-16:
When the joint limit is reached, e.g. arm_7_joint, and the control_mode_adapter switches back from velocity to trajectory mode, switching fails and the switching service returns false, and by false the controller_manager stays in velocity control mode but the joints that were not in the joint limits,  have already switched to trajectory mode while arm_7_joint  is in error state(joint limits) and thus cannot switch.

The desired behavior would be to do not switch any joint, in case one has an error state.

ipa-fmw, 2015-02-09:
@thiagodefreitas: where does this need to be implemented (package)?

ipa-mdl, 2015-02-09:
Will be fixed in the ros-controls stacks, but we need some clean-up on our side as well.

closed 2015-04-29 by ipa-fmw