ros-industrial/ros_canopen #59: [cob4-2] arm moves jerky on "small" joint velocities

ipa-fxm, 2015-01-20:
When sending a ```joint_group_velocity/command``` with joint velocities of 0.01, the arm moves very jerky, i.e. stop-and-go.
When joint velocities of 0.1 are send, the motion is much smoother and continuous.

Solution: Check error tolerances/accuracy of Schunk motor controller

ipa-fxm, 2015-01-20:
a configuration where jerky motion occurs is e.g.(for arm_left):

```
header: 
  seq: 149363
  stamp: 
    secs: 1421784053
    nsecs: 162696290
  frame_id: ''
name: ['arm_left_1_joint', 'arm_left_2_joint', 'arm_left_3_joint', 'arm_left_4_joint', 'arm_left_5_joint', 'arm_left_6_joint', 'arm_left_7_joint']
position: [0.7366860239742865, 1.3486508196010583, -0.2970026788118751, 0.3228335517413911, 0.9281835495031044, 1.5277565074407164, 0.0701447826376521]
velocity: [0.0007853981633974483, 0.007923794804054257, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [9.93e-322, 1.2058886263e-314, 0.0, 5e-324, 0.0, 1.023e-321, 2.592506e-318]
```

jerky motion occurs when commanding the following ```/arm_left/joint_group_velocity_controller/command```

```
rostopic pub /arm_left/joint_group_velocity_controller/command std_msgs/Float64MultiArray "layout:
  dim:
  - label: ''
    size: 0
    stride: 0
  data_offset: 0
data: [0,0.01,0,0,0,0,0]" -r 10
```

A screenshot from the plot from a recorded bagfile:
[image]

ipa-fxm, 2015-01-20:
it starts to get "smoother" for a joint_velocities greater than 0.4
[image]

ipa-fxm, 2015-01-20:
A candump from a motion (in similar but not exactly the same joint_states) using a joint_group_velocity_controller publisher:
```
rostopic pub /arm_left/joint_group_velocity_controller/command std_msgs/Float64MultiArray "layout:
  dim:
  - label: ''
    size: 0
    stride: 0
  data_offset: 0
data: [0,0.01,0,0,0,0,0]" -r 10
```

Thus command should be 0.01 or arm_left_2_joint.

Here is the [candump](https://gist.github.com/ipa-fxm/65e6e2aeda9a309e0f6a)

thiagodefreitas, 2015-01-21:
@ipa-fxm Very nice and detailed description :+1:  

As suspected, this is related to the Schunk internal controller.

According to the candump:
* There is no mode change in between
* No device state change
* Continuous velocity is commanded at the CAN level

Could reproduce the behavior with a 0.01 velocity commanded using the Schunk MTS tools.

I have played a little with the Proportional(e.g Kr_speed, Kr_current) and Integral(Ti) values of the cascaded internal Schunk controller. Especially with Ti to try to reduce the oscillation.

At cob4-2, there is also an additional challenge, as when the robot moves the arm down, the system needs to make a very high compensation of the counter-force, what I could not achieve on those modifications for such small velocity.

thiagodefreitas, 2015-01-21:
Just as an update. 

After a discussion with the hardware team, they affirmed that this velocity is too low. This reaches the limitation from the Hall Sensor and the commutation controller.

ipa-fmw, 2015-01-21:
so what's the solution? we should probably throw some warnings on the terminal if the commanded velocity is below a certain limit. This should be reflected in diagnostics too. @ipa-mdl: what's your suggestion?

ipa-mdl, 2015-01-21:
This is no driver issue, but a controller issue.
We could add a parameter to warn about it, but this is no solution for the problem.
(and currently motor level diagnostics are missing completely..)
First the internal controllers should be tuned, maybe Schunk can help with it or at least confirm the slow velocity limit.
In addition the high level controllers needs to be adjusted to avoid the low ranges, if possible.

ipa-fxm, 2015-01-22:
When commanding 
```
rostopic pub /arm_left/joint_group_velocity_controller/command std_msgs/Float64MultiArray "layout:
  dim:
  - label: ''
    size: 0
    stride: 0
  data_offset: 0
data: [0,0,0,0,0,0,0.1]" -r 10
```
to arm_left_7_joint and plotting the actual velocity from the joint_states topic it looks like this

[image]

The deviation is about +/-0.03 for a joint without much load (it's the last joint!)

So we definitely should investigate/try whether the motor controller settings could be optimized!

thiagodefreitas, 2015-01-22:
For the moment, all the modules (ERB and PRL+), come with the same default controller parameters, as the image below shows.

[image]

ipa-fmw, 2015-02-09:
@thiagodefreitas did some tuning of the firmware controllers. Should be better now. @ipa-fxm and @acuda: please test again.

ipa-fmw, 2015-03-16:
seems to be better now. currently no more firmware tuning. I'll close this.

closed 2015-03-16 by ipa-fmw