ros-industrial/ros_canopen #60: no velocities in joint_states topic

ipa-fxm, 2015-01-21:
There are no entries in the velocity section of the ```joint_states``` topic 

```
---
header: 
  seq: 8573
  stamp: 
    secs: 1421858827
    nsecs: 323728640
  frame_id: ''
name: ['arm_left_1_joint', 'arm_left_2_joint', 'arm_left_3_joint', 'arm_left_4_joint', 'arm_left_5_joint', 'arm_left_6_joint', 'arm_left_7_joint']
position: [0.6421589916862737, 1.1260340735091816, -0.343812409350363, -0.00757472895365539, 0.7335095247356569, 0.9614320717535964, -0.3070208687183225]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [0.0, 2.89e-321, 2.895e-321, 1.040420863149002e+224, 0.0, 1.8146065985459775e+237, 0.0]
---
```

not even in velocity mode

ipa-mdl, 2015-01-21:
According to the [candump](https://gist.github.com/ipa-fxm/65e6e2aeda9a309e0f6a) for issue 59 the velocity is zero in the messages.

ipa-fxm, 2015-01-23:
Don't know what happend when we tried to listen to velocities in the joint_states topic the other day...but everything seems to be ok and working....sorry...closing!

closed 2015-01-23 by ipa-fxm