ros-industrial/ros_canopen #61: driver crashes (sometimes) due to switching controllers in velocity mode

ipa-fmw, 2015-01-23:
a) this one works:
roslaunch cob_bringup robot.launch
init
rosservice call /arm_left/controller_manager/switch_controllers (stop_controllers = joint_trajectory_controller, start_controllers = joint_group_velocity_controller)

b) this one crashes (sometimes, but very often):
roslaunch cob_bringup robot.launch
init
rostopic pub /arm_left/joint_group_velocity_controller/command std_msgs/Float64MultiArray ...

The difference between a) and b) is that in a) we're only switching controllers and in b) we're switching controllers while publishing velocities. There are two subscribers for the velocity topic. So the crash might be caused by a race condition when a callback in the driver is executed before the callback and switch in the cob_control_mode_adapter has finished successfully.
@thiagodefreitas: can you please investigate that on monday?
If that's true, the driver should ignore all velocity inputs as long as not all modules are switched successfully to velocity mode (initiated by the cob_control_mode_adapter). 

@ipa-mdl: FYI

ipa-mdl, 2015-01-27:
problem was already fixed in https://github.com/ipa320/ros_controllers/commit/97f887baa2efe1bd328050c85d374186cebd5865,
but the ipa320 overlay was removed from the robot user and the fix was not merged upstream.

However, ForwardJointGroupCommandController is not thread-safe and needs to be adjusted,
otherwise commmands could be mixed up from different input messages.

The ipa320 is just an intermediate fix for preventing the crashes.

ipa-fxm, 2015-01-28:
There is currently a similar discussion [here](https://github.com/ros-controls/ros_controllers/issues/157) with a pending PR [here](https://github.com/ros-controls/ros_controllers/pull/160)

But as just discussed with @ipa-mdl there is more to that and it is present in other controllers, too... the respective PRs will be updated or superseded...

commit 2e6891d8 (2015-02-10): proper locking for hardware interface switch (might fix #61)
closed 2015-02-10 by ipa-fmw