ros-industrial/ros_canopen #62: driver crashes during init or when EM stop is pressed

thiagodefreitas, 2015-01-27:
When it happens? When initializing some modules, or when the EM stop is pressed, for all of them.

This information was collected during the tests with the base.

Affected HW: 
* It happens for the chain of 6 Elmos, configured with 1Mbit.
* For the individual Elmo modules

The console output is:
 ```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >'
  what():  std::exception
[base/base_driver-1] process has died [pid 4985, exit code -6, cmd /home/thiago/indigo/devel/lib/canopen_motor_node/canopen_motor_node joint_states:=/joint_states __name:=base_driver __log:=/home/thiago/.ros/log/76fa58e8-a62f-11e4-bd4f-f01faf6235e5/base-base_driver-1.log].
log file: /home/thiago/.ros/log/76fa58e8-a62f-11e4-bd4f-f01faf6235e5/base-base_driver-1*.log 
```
After a small investigation, the Exception comes from the [sdo.cpp](https://github.com/ipa320/ros_canopen/blob/indigo_dev/canopen_master/src/sdo.cpp#L413) file.

I could not solve the problem using the solution from https://github.com/ipa320/cob_robots/commit/9131ed8371fe67410b4d26803bf49c86039659ef .

As some additional info, the following generated [candump](http://pastebin.com/kDrySkTS) can be used.

thiagodefreitas, 2015-01-27:
[Raw candump] (http://pastebin.com/jjKaQ8WL)

thiagodefreitas, 2015-01-28:
For one module, it seems to work fine after https://github.com/thiagodefreitas/ros_canopen/pull/6.

When more modules are started, the following error occurs at **init**.

``` 
process[base/base_driver-1]: started with pid [17244]
[ INFO] [1422439150.213281815]: Initializing XXX
[ INFO] [1422439150.213506333]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1422439150.213612993]: Current state: 2 device error: system:0 internal_error: 0 (OK)
RPDOs: 2
TPDOs: 3
error: 4
[ INFO] [1422439154.164757322]: Current state: 2 device error: system:0 internal_error: 4 (controller problems;)
[ INFO] [1422439154.164868868]: Current state: 1 device error: system:0 internal_error: 4 (controller problems;)
ID: 4
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >'
  what():  std::exception
[base/base_driver-1] process has died [pid 17244, exit code -6, cmd /home/thiago/indigo/devel/lib/canopen_motor_node/canopen_motor_node joint_states:=/joint_states __name:=base_driver __log:=/home/thiago/.ros/log/9658a2b4-a6d2-11e4-b601-f01faf6235e5/base-base_driver-1.log].
log file: /home/thiago/.ros/log/9658a2b4-a6d2-11e4-b601-f01faf6235e5/base-base_driver-1*.log

```

The following [candump] (http://pastebin.com/j4gvLnCg) was recorded until the crash point.

thiagodefreitas, 2015-01-28:
Updated [dump] (http://pastebin.com/2xNKiLVy)

ipa-mdl, 2015-01-28:
Tested on test rigs and cob4, works for kernel version 3.11, but not 3.13.

thiagodefreitas, 2015-01-29:
I still have the same error.

Until now, I have tested the following Kernels with the Socketcan driver:
3.11.10-03 :x:
3.13.0-32 :x:
3.13.0-34 :x:
3.13.0-44 :x:

thiagodefreitas, 2015-01-30:
It is complicated to associate a Kernel or USB version to the issue. As I could properly run the driver using a ThinkPad T430, Kernel 3.31.0-44 and USB 3.0 and 2.0.

A NUC PC has exactly the same behavior as my Notebook.

```
process[rig2/rig2_controller_spawner-8]: started with pid [23039]
[ INFO] [1422613568.275873763]: Initializing XXX
[ INFO] [1422613568.276174032]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1422613568.276415228]: Current state: 2 device error: system:0 internal_error: 0 (OK)
RPDOs: 2
TPDOs: 3
Homing not started
Homing succesful
Propertly initialized module
Enter mode1
[ INFO] [1422613575.307061005]: Initializing XXX
[ INFO] [1422613575.307348616]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1422613575.307516254]: Current state: 2 device error: system:0 internal_error: 0 (OK)
error: 4
error: 4
[ INFO] [1422613575.311141075]: Current state: 2 device error: system:0 internal_error: 4 (controller problems;)
[ INFO] [1422613575.311154028]: Current state: 2 device error: system:0 internal_error: 4 (controller problems;)
[ INFO] [1422613575.311216254]: Current state: 1 device error: system:0 internal_error: 4 (controller problems;)
[ INFO] [1422613575.311235817]: Current state: 1 device error: system:0 internal_error: 4 (controller problems;)
ID: 4
ID: 4
error: 4
error: 4
ID: 4
ID: 4
[ERROR] [1422613576.319097433]: /home/robot/git/care-o-bot/src/ros_canopen/canopen_master/src/sdo.cpp(413): Throw in function void canopen::SDOClient::wait_for_response()
Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >
std::exception::what: std::exception

error: 4
error: 4
ID: 4
ID: 4
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >'
  what():  std::exception
[rig2/driver-6] process has died [pid 23004, exit code -6, cmd /home/robot/git/care-o-bot/devel/lib/canopen_motor_node/canopen_motor_node joint_states:=/joint_states __name:=driver __log:=/home/robot/.ros/log/64404c98-a86a-11e4-95f3-eca86bffd49a/rig2-driver-6.log].
log file: /home/robot/.ros/log/64404c98-a86a-11e4-95f3-eca86bffd49a/rig2-driver-6*.log

```

thiagodefreitas, 2015-01-30:
Using the peak-can driver(not the Kernel Version, the issue is very similar), For this specific run, the error code from the [controller problem](https://gitorious.org/linux-can/can-utils/source/77de10356fe0eca7bce73e9c3ac4ed32edfc10e5:include/socketcan/can/error.h#L36-40), although for another it was 20.

For a candump with -dxta, the result is:

```
(1422613575.299015)  can0  RX B E  5D5   [8]  4F 01 10 00 00 00 00 00
 (1422613575.299306)  can0  TX B E  255   [6]  0F 00 00 00 30 F3
 (1422613575.299446)  can0  TX B E  080   [0] 
 (1422613575.300644)  can0  RX B E  1D5   [3]  37 06 01
 (1422613575.300917)  can0  RX B E  181   [3]  37 06 01
 (1422613575.301160)  can0  RX B E  381   [8]  00 00 00 00 00 00 00 00
 (1422613575.301386)  can0  RX B E  3D5   [8]  00 00 00 00 00 00 00 00
 (1422613575.307896)  can0  TX B E  655   [8]  40 01 10 00 00 00 00 00
 (1422613575.308622)  can0  TX B E  000   [2]  81 01
 (1422613575.308963)  can0  RX B E  5D5   [8]  4F 01 10 00 00 00 00 00
 (1422613575.309097)  can0  TX B E  255   [6]  0F 00 00 00 30 F3
 (1422613575.309149)  can0  TX B E  080   [0] 
 (1422613575.310373)  can0  RX B E  1D5   [3]  37 06 01
 (1422613575.310640)  can0  RX B E  181   [3]  37 02 01
 (1422613575.311031)  can0  RX - -  20000004   [8]  00 04 00 00 00 00 00 00   ERRORFRAME
 (1422613575.311586)  can0  RX B E  381   [8]  00 00 00 00 00 00 00 00
 (1422613575.311811)  can0  RX B E  3D5   [8]  00 00 00 00 00 00 00 00
 (1422613575.311031)  can0  RX - -  20000004   [8]  00 04 00 00 00 00 00 00   ERRORFRAME
 (1422613575.311031)  can0  RX - -  20000004   [8]  00 04 00 00 00 00 00 00   ERRORFRAME
 (1422613579.021705)  can0  RX B E  701   [1]  00

```

@ipa-mdl : Maybe a RESET_NODE instead of  a RESET_COMMUNICATION during init could help on this issue. What do you think?

thiagodefreitas, 2015-01-30:
For the 3.16.0-29-generic Kernel, using the PCAN, there was some succesful runs, but it also gave quite frequently the previously mentioned error.

It seems to work using an ESD/USB-2 dongle. We need maybe to clarify that with Peak and/or Elmo.

thiagodefreitas, 2015-01-30:
Just as an update: The same happens for the low-latency Kernels.

ipa-fmw, 2015-02-13:
this is related to #67

closed 2015-02-13 by ipa-fmw