ros-industrial/ros_canopen #67: difference between ELMO and SCHUNK

ipa-fmw, 2015-02-12:
in canopen_master/src/node.cpp we currently need

for SCHUNK:
```
        reset();
```

and for ELMO (if more than one module is on the bus):
```
        reset_com();
        emcy_.recover();
```

@thiagodefreitas, @ipa-mdl: please think about a solution which works for both

thiagodefreitas, 2015-02-27:
For running the test on the cob4-6, I had to do the following changes:

  * canopen_master/src/node.cpp
  void Node::init(LayerStatus &status){
     nmt_listener_ = interface_->createMsgListener( can::MsgHeader(0x700 + node_id_), can::CommInterface::FrameDelegate(this, &Node::handleNMT));
 
     sdo_.init();
     try{
        reset_com();
        emcy_.recover();
         prepare();
     } 
     ...
  * canopen_master/src/emcy.cpp
      void EMCYHandler::read(LayerStatus &status)
      {
      /*
           if(error_register() & (~32)){ // mask device specific error
               status.error("Node has emergency error");
           }
          */

ipa-mdl, 2015-02-27:
This disables the error detection (incl. emergency stop) completely!
The EMCY handing and especially the reset needs some more work done, I will try to find some time for it next week.

ipa-fmw, 2015-04-29:
current state should work with both. needs to be tested

ipa-mdl, 2015-06-16:
Current implementation should work for most devices, if not specific issues should be filed.

closed 2015-06-16 by ipa-mdl