ros-industrial/ros_canopen #68: refactore 402 package

ipa-fmw, 2015-02-13:
- timeout for homing
- reflect motor state machine (switch mode, set velocities, ...)
- encapsulate in base class

thiagodefreitas, 2015-02-27:
@ipa-mdl , @ipa-fmw I have started with the refactor.

For that, I have started refactoring the state machines, what I consider the kernel of the 402 and the most important part to work properly.

I took the decision to change the SMs to use the boost Meta State Machine (MSM), where I can get a better view of the actions and the guards and to use the transition table, where I and anyone else in the future could have a better view of what is really happening. And also have more reliability concerning the state transitions, the mode switching and all the internals. Please, see, for a general idea:

https://github.com/thiagodefreitas/ros_canopen/blob/refactor_sm/canopen_402/include/canopen_402/state_machine.h#L165-L193

I am implementing the 402 SM and the sub-machines, as well as I want to implement a higher level SM (ON, OFF, etc...), so it is not everything online.

This should take quite some more time and effort, and in my opinion is a relevant change. If you have something against the approach, please inform me as soon as possible, so that I may concentrate the efforts on the other aspects.

Thanks.

ipa-mdl, 2015-02-27:
This sounds great, the 402 proflie is rather complicated and a clean state machine implementation would improve the robustness and maintenance.
I am a little bit concerned with the effort, but the benefits may be worth it.

ipa-fmw, 2015-03-09:
this is a critical issue for https://github.com/ipa320/ros_canopen/milestones/ROS-I%20release

@thiagodefreitas when do you think this will be finished as it is blocking us with the ROS-I release? Where can I find the modifications you already did? what is still missing?

closed 2015-06-16 by ipa-mdl