ros-industrial/ros_canopen #74: Add support for hex numbers in eds file

mistoll, 2015-03-09:
Some vendors use hex numbers for "Unsigned#" data types in eds files. It would greatly enhance the compatibliity if the parser would accept these values. 
And according to CanOpen 306 chapter 4.3 this should be supported.

Example EDS:
  [MandatoryObjects]
  SupportedObjects=0x03
  1=0x1000
  2=0x1001
  3=0x1018

For the whole file see: http://en.nanotec.com/products/1604-pd4-c-high-pole-plug-drive-dc-servo-motor/

ipa-mdl, 2015-03-09:
Thanks for the hint!

I have implemented a fix that should read all numbers with hex support (for parameters this was already supported).
In addition the nanotec EDS contains octet strings without the leading 0 (as required by 306), I have added support for it as well.
Furthermore I patched the IP mode handling, that prevented the driver to start in my test.

The patches have not been merged upstream yet, since I need to test them first with our devices.

ipa-fmw, 2015-03-09:
fixed with #75

mistoll, 2015-03-09:
Great, that was quick.
Unfortunately I have some trouble builiding the indigo-devel branch. So I cannot test it.

The pull request (https://github.com/ipa320/ros_canopen/issues/75) is orthogonal to this issue

ipa-fmw, 2015-03-09:
@mistoll: can you share details on your build errors? e.g. using https://gist.github.com/ and then post a link here?

mistoll, 2015-03-09:
@ipa-fmw: Sure, here it is: https://gist.github.com/mistoll/2c2e35740f8f19c77d1c
Looks like some boost issue. Boost version is 1.54.0.

ipa-mdl, 2015-03-09:
I was not able to reproduce these errors on a clean Ubuntu 14.04 64bit system.
Which system do you use?
Do you have any special compiler settings? (E.g. -std=c++11 does not work).
Have you tried to build it from a clean workspace?

mistoll, 2015-03-10:
System is Ubuntu 14.04.2 32bit on a 64-bit machine.
Same with clean workspace, no special compiler settings.

ipa-mdl, 2015-03-10:
I was able to reproduce these errors in a 32 bit VM, the fix was easy :)

PR #80 (2015-04-07): Bug fixes and tweaks
tested on test rigs  elmo1 and schunk, partially on cob4-6.

Updates:
* closes #74 
* includes #76 
* new struct-based formats (backwards compatiple with lists)
* producer and and consumer heartbeats
* new PDO mapping, new readable.py: usage readable.py elmo_mapping or readable.py schunk_mapping
* disabled loopback, removed loopback listener
* added string serializers
* simplified EMCY handling
* support for silence time before sync
* removed some timeout-related crashes
* error logging
* new configs for test rigs
* refactored Layer system

Known issue:
* recover might be  [... cut]
closed 2015-04-07 by ipa-mdl