ros-industrial/ros_canopen #87: Second io_service processing thread in socketcan_interface/asio_base.h causes packet reordering at high CAN data rates

sjs329, 2015-04-16:
Line 86 in https://github.com/ipa320/ros_canopen/blob/indigo_dev/socketcan_interface/include/socketcan_interface/asio_base.h starts an additional thread to call io_service::run() even though it gets called in the current thread at line 91. The second thread does not seem to provide much in the way of performance enhancements, and it causes CAN frames to get dispatched out of order occasionally when the data rate is very high. 

This was seen running a CAN device that outputs ~80 messages at 25 Hz each (total of 2000Hz). Simply commenting out line 86 fixed the reordering issue. Without that line, boost/thread/thread.hpp no longer needs to be included.

ipa-mdl, 2015-04-16:
Thanks for the report!
The second thread was meant to decouple the read from the processing which may block.
I don't want to remove the second thread for now since this might break the other layers.

However, I have prepared a patch that runs the processing through a strand to keep the original order: https://github.com/ipa-mdl/ros_canopen/tree/can_ordering
Can you please that version?

sjs329, 2015-04-16:
I appreciate the rapid reply! 
Using the strand is a great solution and appears to fix the reordering issue. Thank you very much!

closed 2015-04-16 by sjs329