ros-industrial/ros_canopen #88: switch to std_srvs/Trigger instead of cob_srvs/Trigger

ipa-fmw, 2015-04-17:
std_srvs/Trigger is in the release pipeline already, see ros-controls/ros_control#197.

we'll need to change the dependencies to use std_srvs/Trigger instead of cob_srvs/Trigger

ipa-mdl, 2015-04-17:
In addition all simple_script_sever-related interfaces must be adapted, including other component drivers.

ipa-fxm, 2015-04-17:
I can do the non-ros_canopen part...

ipa-fxm, 2015-04-18:
I did this for all bringup repositories:
 - https://github.com/ipa320/cob_command_tools/pull/94
 - https://github.com/ipa320/cob_control/pull/31
 - https://github.com/ipa320/cob_driver/pull/217
 - https://github.com/ipa320/cob_manipulation/pull/51
 - https://github.com/ipa320/cob_robots/pull/293
 - https://github.com/ipa320/schunk_modular_robotics/pull/135

All other repos do not use Trigger - or did I miss something?

I leave ros_canopen for @ipa-mdl 
What about ipa_canopen? For compatibility reasons?

I'd adjust ```cobvid``` and ```hmi_2015``` on Wednesday!

Any other packages where we need to change the service?

ipa-fmw, 2015-04-21:
good work @ipa-fxm! I can't think of any other repository.

But I'd like to remove cob_srvs/Trigger because it add no additional value compared to std_srvs/Trigger. By removing cob_srvs/Trigger we assure that it pops up if it is still used anywhere else.

PR #90 (2015-06-16): migrated to std_srvs/Trigger
Closes #88 
Should be merged after https://github.com/ros/rosdistro/pull/7964
closed 2015-06-16 by ipa-mdl