ros-industrial/ros_canopen #92: calibration rising

jobstvogt, 2015-04-28:
I think the calibration rising values from the urdf are ignored at the moment...

ipa-fxm, 2015-04-28:
Yes, as it seems not to be supported by the urdfdom library anymore...(compared to the ros-specific version that was used until groovy/hydro)

ipa-mdl, 2015-04-28:
According to https://github.com/ros/urdfdom_headers/blob/master/urdf_model/include/urdf_model/joint.h#L207 it is an "Unsupported Hidden Feature"
In addition the semantics are not really clear.

jobstvogt, 2015-04-29:
But it was implemented in ipa_canopen_ros.cpp:
```
/// Get offsets out of urdf model
if(!model.getJoint(jointNames[i].c_str())->calibration)
{
    ROS_ERROR("Parameter calibration could not be found in the URDF contents.");
    n.shutdown();
    exit(1);
}
else if(!model.getJoint(jointNames[i].c_str())->calibration->rising)
{
    ROS_ERROR("Calibration has no rising attribute");
    n.shutdown();
    exit(1);
}
Offsets[i] = model.getJoint(jointNames[i].c_str())->calibration->rising.get()[0];
```
Do you have an idea how to get the offsets of the calibration-routine to work in ros_canopen?

ipa-mdl, 2015-04-29:
There is no need for offsets if you home the hardware.
The offsets in ipa_canopen interfered with the hardware limits and should better be zero anyway.
We do not support custom offsets in ros_canopen by now. Why do you need them?

jobstvogt, 2015-04-29:
I have changed the calibration-routine that it optimizes also the joint-offsets and i get much better calibration results with this. I have this arm-offsets:
```
	<property name="offset_arm_1_ref" value="0.0"/>
	<property name="offset_arm_2_ref" value="-0.1607150542"/>
	<property name="offset_arm_3_ref" value="0.0213738687"/>
	<property name="offset_arm_4_ref" value="0.0059519017"/>
	<property name="offset_arm_5_ref" value="-0.0229626904"/>
	<property name="offset_arm_6_ref" value="0.0013964296"/>
	<property name="offset_arm_7_ref" value="0.0005167186"/>
```
Maybe i can write these values to the hardware config with the Schunk Motion Tool...

ipa-mdl, 2015-04-29:
We currently do not support cob3 with indigo/ros_canopen.

closed 2015-04-29 by ipa-mdl

jobstvogt, 2015-04-29:
What about cob3-9? I was told that cob3-9 is working fine in Indigo and it is possible to upgrade. Indeed there is an installation manual for cob3/indigo in ipa320/setup...

ipa-fmw, 2015-04-29:
cob3-9 uses a preliminary indigo version of the cob repositories which is not cleaned up to be used for all robots. We never recommended anybody else to upgrade to indigo at this time.

moriarty, 2016-02-24:
What is the current status of this issue? 

You say you do not recommend upgrading to Indigo, yet the packages have been released... if Indigo isn't supported this should be made more visible. 

We're currently in the middle of upgrading cob3-1.

ipa-mdl, 2016-02-24:
The cob3-1 does not have any `ros_canopen` components.

> What is the current status of this issue?

This won't be fixed. I have added the label now.

> yet the packages have been released... 

The released version of `ros_canopen` is outdated and lacks important features (next release: https://github.com/ros-industrial/ros_canopen/milestones/Final%20indigo%20release)

> You say you do not recommend upgrading to Indigo

We don't have a cob3-1 config for indigo, this is quite visible (https://github.com/ipa320/cob_robots/tree/indigo_dev/cob_hardware_config/).

moriarty, 2016-02-24:
Sorry, we had the same errors in similar section of code, you are correct it wasn't ```ros_canopen```.

We were able to fix the issue and have submitted a pull request to ipa320/schunk_modular_robotics#159