ros-industrial/ros_canopen #97: pg70 gripper driver isn't starting anymore

jobstvogt, 2015-06-18:
The gripper driver crashes at startup because of the pos and vel_unit_factor in the yaml-file...

ipa-mdl, 2015-06-18:
Please don't file device specific issues in this repository, it is just for the driver, not for configurations.
However, the crash is intended. The unit factors aren't supported anymore in favor of the muparser-based conversion functions (#95).
The defaults (mdeg<->rad) should work for the PG70 plus if you set it to mdeg units.
Long story short: Your configuration is outdated, please open an issue or ever better a PR in the respective repository.

closed 2015-06-18 by ipa-mdl

jlarraez, 2015-09-24:
I know this topic is close and this is not the right place for answering this but, how I manage to use the pg70 gripper with the ros_canopen? Actually I am able to move the arm but dont know how to implement the gripper. There is any tutorial about how to configure it?

ipa-mdl, 2015-09-24:
We currently do not have a configuration tutorials available (#29), but the configuration is listed here: http://wiki.ros.org/canopen_motor_node

The PG70 plus (**!!**) uses the same firmware and interfaces as the LWA4 modules, so it should be possible to use the LWA4 packages as a template.

It is even possible to start two instances for one CAN bus, but this feature is currently not robust (#71)

ipa-fxm, 2015-09-24:
I think you could have a look at cob3-9 which uses a pg70. However, as we do not have this robot around, no guarantee that this is a working config:
 - https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_bringup/robots/cob3-9.xml#L90-L98
 - https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob3-9/config/gripper_driver.yaml
 - https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/cob3-9/config/gripper_controller.yaml

jlarraez, 2015-09-25:
Hi!. Thanks for the fast reply!. I will check the info and post the advance I manage to do!

jobstvogt, 2015-09-29:
Hi, i have a working config for pg70.

You can find the config files here:
* https://github.com/jobstvogt/cob_robots/blob/indigo_dev/cob_hardware_config/cob3-8/config/gripper_driver.yaml
* https://github.com/jobstvogt/cob_robots/blob/indigo_dev/cob_hardware_config/cob3-8/config/gripper_controller.yaml
* https://github.com/jobstvogt/cob_robots/blob/indigo_dev/cob_bringup/robots/cob3-8.xml

The important thing is the conversion stuff in gripper_driver.yaml and you need the correct netdev can drivers.

jlarraez, 2015-09-29:
Hi thanks guys! I have already finished the config files for the PG70 and the schunk Lwa4p.
They are all here:
https://github.com/jlarraez/schunk_lwa4p

ipa-fxm, 2015-10-01:
Could you guys provide a PullRequest in order to play back your updated config files to both `ipa320/cob_robots` and `ipa320/schunk_robots`?