ros-industrial/ros_canopen #99: Trigger new release

ipa-fmw, 2015-07-02:
We're breaking the OSRF buildfarm at the moment. A new release will fix that.

Code is already pushed to the indigo_release_candidate branch. Next step is to release with bloom and push to the release repository.

@shaun-edwards: please add the release repository (https://github.com/ros-industrial/ros_canopen-release) to the admin team (https://github.com/orgs/ros-industrial/teams/canopen-admin/repositories) so that we can continue with the release. Thanks.

gavanderhoorn, 2015-07-02:
There is a special organisation that hosts all the `*-release` repositories: https://github.com/ros-industrial-release.

ipa-fmw, 2015-07-02:
thanks @gavanderhoorn, good point. Unfortunately only @shaun-edwards can transfer the repo now because I don't have access to it. 

@shaun-edwards: please transfer again to ros-industrial-release and add a team with push access to ros_canopen-release which lists @ipa-mdl and @ipa-fmw. Thanks.

ipa-fmw, 2015-07-02:
just found out that I am already in the release group. @shaun-edwards just needs to transfer the repository

shaun-edwards, 2015-07-02:
I'm not getting transfer emails. @ipa-fmw, you are an owner of ROS Industrial release. You should have all the access you need.

ipa-fmw, 2015-07-02:
for ros-industrial-release yes, but not for ros-industrial.

This time you have to trigger the transfer because the repo belongs to ros-industrial alrealy (to which I don't have admin access). Just go to https://github.com/ros-industrial/ros_canopen-release/settings and press the transfer button

shaun-edwards, 2015-07-02:
Ok...all fixed.  I transfered the repo.  I also gave you admin rights (and some others as well) so you should be able to do this in the future.  That way you aren't waiting on me.

ipa-fmw, 2015-07-03:
Perfect. Thanks for your effort. The release is in the pipeline: https://github.com/ros/rosdistro/pull/8840

closed 2015-07-03 by ipa-fmw