ros-industrial/ros_canopen #102: Investigate socketcan_interface queue overflows

ipa-mdl, 2015-07-06:
The socketcan txqueue might run full on fast updates, socketcan_interface does not handle this properly.

A temporary fix is to increase the queue len:
```
sudo ip link set <NETWORK> txqueuelen 15 # or even higher, default: 10
```
socketcan_interface should detect the buffer overrun and handle it properly.
Problem: In most case the error is caught in the following read and not in the actual write.
Use async write instead?

ipa-fxm, 2015-10-30:
Can this be integrated into https://github.com/ros-industrial/ros_canopen/blob/indigo-devel/canopen_test_utils/scripts/prepare.sh?

ipa-mdl, 2015-10-30:
No, because it is quite cumbersome to do argument parsing in bash. 
This script will be removed sooner or later..

I recommend to put the following in `/etc/network/interfaces`:
```
allow-hotplug can0
iface can0 can static
    bitrate 500000
    up ip link set $IFACE txqueuelen 20
```

I have updated the wiki as well. (http://wiki.ros.org/socketcan_interface#Initialize_NIC)