ros-industrial/ros_canopen #103: Move socketcan_interface into dedicated repostory

ipa-mdl, 2015-07-07:
Might be reasonable for CAN-only development

JWhitleyAStuff, 2019-03-13:
Do you know if anyone is using socketcan_interface independently (anyone outside of this repository)?

Timple, 2019-03-14:
We do for some of our robots. For prototyping it is easier to use the can_msgs than to open up a socket connection.
But we don't mind it being in this repo actually.

ipa-mdl, 2019-03-15:
For the released version it does not matter, but for developement t might be easier to just have the CAN-related packages (can_msgs, socketcan_interface, socketcan_bridge) without the ros_control dependencies.

samiamlabs, 2019-03-15:
I used socketcan_interface independently from this repo for a robot a couple of years ago. The hardware I was interfacing with was only partially using CANopen. Made a custom ros_control hardware interface.

I agree that having it in this repo is ok, but separating would be nice for development.

JWhitleyAStuff, 2019-03-15:
@ipa-mdl Can you create the new repo? Do we want to do this for just Melodic or back-port it for others?

ipa-mdl, 2019-03-18:
> Can you create the new repo? 

Actually, I can't..

>  Do we want to do this for just Melodic or back-port it for others?

I would do it for the ROS2 port and ROS-N.
Not sure about melodic.

@gavanderhoorn: What do you think about splitting the repository?

gavanderhoorn, 2019-03-18:
@ipa-mdl wrote:
> @gavanderhoorn: What do you think about splitting the repository?

Is the desire mostly to be able to use `can_msgs` without any of the other packages, or really to have access to `socketcan_interface` only (with `can_msgs`)?

If the former: I'd take `can_msgs` out of this repository. If the latter: is it just about dependencies, or also about compilation times?

@ipa-mdl: what is the status of the packages in this repository? Mostly maintenance, or active development? Splitting repositories will complicate PR management as you have to sync everything to make sure they get merged in the proper order (for instance).

ipa-mdl, 2019-03-18:
I would keep `can_msgs`, `socketcan_interface` and `socketcan_bridge` together. to split into CAN and CANopen set.

> If the latter: is it just about dependencies, or also about compilation times?

Mostly dependencies. There are 2 groups of users (CAN data transfer and ros_control-based HW), so it might make sense to separate them.

>what is the status of the packages in this repository? Mostly maintenance, or active development?

I would say maintenance. But there is some ROS2 porting in the making, which is the most compelling reason to split.. (ros2_control is not yet ready).

> Splitting repositories will complicate PR management as you have to sync everything to make sure they get merged in the proper order (for instance).

Normally, the two sets don't have a big overlap, I don't see difficulties here.
I still plan to maintain both.

> would you want to maintain commit history for socketcan_interface? 

I guess I would go for `git filter-branch`.
For now I just need a new (empty) repositories ("socketcan_interface"?) with the admin group..

gavanderhoorn, 2019-03-18:
> > would you want to maintain commit history for socketcan_interface?
> 
> I guess I would go for `git filter-branch`.

Hm. The whole repository is only 2.6 MB, with 2 MB of history. `filter-branch` will rewrite all of history, invalidating all references to commit hashes in the process. Personally I would go for a mirror-and-delete, but it's up to you.

I'm not really convinced that ROS2 development is a good argument for splitting repositories, but see [ros-industrial/socketcan_interface](https://github.com/ros-industrial/socketcan_interface).

JWhitleyAStuff, 2019-03-18:
@ipa-mdl If you tell me what you want the repository to be named, I can message the group admins.

ipa-mdl, 2019-03-18:
Thanks @gavanderhoorn!
I am not sure if I want to keep the separate repository, but for now it is a good testbed :)
That's why I will keep the packages in both places!

I will populate `melodic-devel` and `ardent-devel` from @JWhitleyAStuff's respective branches.

> I'm not really convinced that ROS2 development is a good argument for splitting repositories

It is not a justification for the split, but for the timing..

gavanderhoorn, 2019-03-18:
> I am not sure if I want to keep the separate repository, but for now it is a good testbed :)

Oh. Wouldn't testing have worked just as well in a repository on your personal account?

> I will populate `melodic-devel` and `ardent-devel` [..]

Crystal has already seen 3 patch releases, wouldn't it make more sense to target that?

JWhitleyAStuff, 2019-03-18:
@gavanderhoorn and @ipa-mdl - I am targeting crystal and the branch on my repo is crystal-devel.

ipa-mdl, 2019-03-18:
> Oh. Wouldn't testing have worked just as well in a repository on your personal account?

Testing -> testing the new repository structure.
If we see that there is no need to split the repositories, then we could include them here again.

> Crystal has already seen 3 patch releases, wouldn't it make more sense to target that?

Ah, sry, ardent is already EOL. so I would create `bouncy-devel`, which should support bouncy and crystal. This is the same scheme as for all other distros.
If they are not compatible, we might drop bouncy support and go for crystal only.
I have already prepared a `industrial_ci` prototype for ROS2, so it's easy to maintain.

gavanderhoorn, 2019-03-18:
> If we see that there is no need to split the repositories, then we could include them here again.

then I would definitely recommend the mirror-repository-with-delete approach, as it would make reintegrating the two repositories (while maintaining history) a lot easier.

ipa-mdl, 2019-03-18:
> then I would definitely recommend the mirror-repository-with-delete approach

agreed (https://github.com/ros-industrial/socketcan_interface/commit/53764bf9c4d74ca5190a25f56e167c818b5d751f)

gavanderhoorn, 2019-03-18:
Did you use [--mirror](https://help.github.com/en/articles/duplicating-a-repository)?

ipa-mdl, 2019-03-18:
> Did you use --mirror?

No. I don't want to mirror everything, but I will check this flag

gavanderhoorn, 2019-03-18:
You can first mirror everything, then delete the branches you don't want/need. That would result in a repository that is as close to a clone of the source as you can get.

ipa-mdl, 2019-03-18:
@JWhitleyAStuff: I think we should merge #320 before migrating to new repository. I will try to finish it as soon as possible. We can rebase your crystal-devel branch easily, works for bouncy and crystal :)

>  That would result in a repository that is as close to a clone of the source as you can get.

This is not what I intended. For now I would keep the ROS1 packages (up to melodic) here and the ROS2 packages there.

gavanderhoorn, 2019-03-19:
> > That would result in a repository that is as close to a clone of the source as you can get.
> 
> This is not what I intended. For now I would keep the ROS1 packages (up to melodic) here and the ROS2 packages there.

you could still do whatever you want after having duplicated the repository. My point was merely that I'd first get a mirror of `ros_canopen` into `socketcan_interface`, and *then* start pruning and merging changes for the ROS 2 packages.

But nm.