ros-industrial/ros_canopen #110: Though not initialized correctly, init service reports success

mistoll, 2015-08-03:
When a init service call fails a second call to the init service might return successfull while not initializing the driver.

Reproduce like this:
1)
Setup a canopen_motor_node.
 
2)
Somehow generate a SDO error while calling the service /driver/init().
You might just add "reason = 0x08000000; break;" into sdo.cpp, handleFrame method, Upload*Response cases to force this. (I can provide a patch, if required)
This init call fails (service error, not a error message)

3)
Call /driver/init() a second time. This time the service call returns
{ success: True, message '' }
Despite of displaying "Initializing XXX" the driver does not do much. The controller is not loaded. The motor is still not operational.

mistoll, 2015-08-04:
In chain_ros.h RosChain::handle_init only canopen::Exception is caught. For other exceptions the state after the exception might be invalid. Maybe there should be a catch all handler?

ipa-mdl, 2015-08-04:
Good point, I switched to std::exception to catch them as well.
But I don't think that this a fix for the issue, I guess the exception is caught in between and not handled properly.

ipa-mdl, 2015-08-04:
The service call was aborted with an exception becauce https://github.com/ros-industrial/ros_canopen/commit/00a2c205e6698abaec8825666366be58ecb568ce decoupled TimeoutException from canopen::Exception.

ipa-mdl, 2015-08-14:
Should be fixed, can this one be closed?

mistoll, 2015-08-17:
Not yet. The cleanup might throw an exception and leave the drive in an undefined state.

I added a try/catch around it in #121. It seems to sovle the issue. But it might not reflect the expected behavior!

ipa-mdl, 2015-08-18:
Shutdown should never throw. It is a bug if it does, since the clean-up would not finish for all layers.
Which exception do you get? (I bet on the 301 Layer shutdown)
Exception handling must be done at the lowest layer possible, this is currently not implemented for all cases.

This is different to the `init()` function, because a mixed state after a failed init would be cleaned up by `shutdown()` ;)

mistoll, 2015-08-19:
So what do you think of #123? This should fix the issue and does not add any exception handling.
However there might be other catches.

ipa-mdl, 2015-08-19:
I was thinking of catching all exceptions in the Layer base class und pass them as errors, I will prepare a patch for this.
However, I'd really like to know which exception do you get.

mistoll, 2015-08-19:
The exceptions are (with reproduction hack in place):
First exception (init):
> [ERROR] [1439991800.608493932]: ros_canopen/canopen_master/src/sdo.cpp(423): Throw in function void canopen::SDOClient::wait_for_response()
> Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >
std::exception::what: SDO: 1800sub1

Second exception (shutdown)
> [ERROR] [1439991800.613236297]: Exception thrown while processing service call: SDO: 1017sub0

ipa-mdl, 2015-08-19:
Assuming that the response did not take 1s, I guess that the SDO started waiting just after the response was received. This is addressed in #117.

mistoll, 2015-08-20:
#117 does not make a significant difference in my test case. (There's more output "Could not prcoess message. discarded message").  However the second exception does look like a timing issue, because it does not happen every time.
With #124 in place the second exception is caught. Seems to fix the issue (at least for my test case)

ipa-mdl, 2015-08-20:
Can you please provide a new candump?
"Could not prcoess message. discarded message" (note to self: fix typos) will be displayed whenever a message is received while the SDO handler is not waiting for one, i.e. no transmission is in progress.

ipa-mdl, 2015-10-19:
Can this issue be closed?

mistoll, 2015-10-19:
I had no errors for a long time, so yes, let's close it.

closed 2015-10-19 by mistoll