ros-industrial/ros_canopen #111: While initializing Target Velocity might be non zero

mistoll, 2015-08-03:
This assumes PV-Mode but should be valid for other modes, too.

When for the Target Velocity of the motor (on the slave) is non zero, the motor might begin rotating for a short period of time during initialization. This is because the the driver sets the Target Velocity object only after the state Operation Enabled has been entered. This happens independently of the Target Velocity on the master.

The precondition (Target Velocity != 0) might happen during any crash or error condition and cannot be eliminated! 
To reproduce it (hardware dependant!):
1) Start canopen_motor_node (PV-Mode)
2) Init and command the motor to move
3) Stop the driver (Ctrl+C), but do not stop the movement command topic
Now you should have a motor with Target Velocity set to non zero, without a driver running. Continue with:
4) Start canopen_motor_node
5) Call init
Motor should move shortly

IMO this issue is critical, because it's potentially dangerous.

From a more general point of view, this problem could influence every mode. Whenever the motor's object dictionary does not match the dcf default values. But the default values cannot be ensured by a previous driver instance, because a crash might produce an unpredictable motor state. IMO setting each critial object dictionary entry on initialization is not viable solution. I'd suggest to reset each node (NMT reset message) before initializing the node. However that would involve a lag because the node needs to boot.

ipa-mdl, 2015-08-03:
We had do remove the deep reset from the driver since it lead to CAN bus problems for Elmo devices.
It can be added again (as an option) after #104 was implemented.

I agree that this issue is critical, I propose to set the Halt bit before the motor is enabled.
This is the default behaviour at start-up, but is currently done shortly (1 sync interval) after the initialization. This delay does not occur on recover.

Can you please test https://github.com/ipa-mdl/ros_canopen/tree/halt-at-init?

mistoll, 2015-08-04:
An optional deep reset would be great.

The branch https://github.com/ipa-mdl/ros_canopen/tree/halt-at-init seems to solve the original problem. But while testing it I experiences an even worse problem making it hard to reproduce the problem. Though I managed to run the reproduction steps once, without seeing any motor movement.

It might not be related to this issue, but here's what I experienced: When doing the same steps as above to reproduce the original problem, on Step 5) one of the motors (there are two in my setup) starts to rotate at a high speed (maybe maximum) for a view Seconds. Then the motor stops and the driver reports:
> Slave timeout
> [ERROR] [1438683069.176670340]: not operational; not operational; not operational; notify failed
> Slave timeout
The service client reports:
> success: False
> message: Transition timeout; Could not enter homing mode
Homing method is set to 0, so there should not be any homing at all.

ipa-mdl, 2015-08-04:
Okay, I will try the halt-init with our devices and merge it afterwards.

Homing mode is entered at start-up if homing is supported.
If homing method is not zero, the procedure is started.
However, according to your output homing mode was not even entered, so no mode was selected and the halt bit should have been set.

It is hard to tell what happened without a candump.

Do you use synchronous PDOs?

ipa-mdl, 2015-08-04:
The halt-at-init branch does not work with the latest Schunk firmware, the Halt bit prevent the transition to OperationEnabled.
The 402 standard demands that all setpoints are clear on that transition.
Did you map the control word and the velocity target to the same RPDO?

ipa-mdl, 2015-08-14:
The false homing seems to be a problem with the motor controller that started the motion before it was switched on.

mistoll, 2015-08-17:
This issue is fixed with #112. The homing problem not related to this issue (seems to be a hardware failture).

closed 2015-08-17 by mistoll