ros-industrial/ros_canopen #128: Building in Indigo

droter, 2015-09-01:
Hi,

When I try to build the package I am getting this error:

-- +++ processing catkin package: 'canopen_motor_node'
-- ==> add_subdirectory(ros_canopen/canopen_motor_node)
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "hardware_interface" with any of the following names:

    hardware_interfaceConfig.cmake
    hardware_interface-config.cmake

  Add the installation prefix of "hardware_interface" to CMAKE_PREFIX_PATH or
  set "hardware_interface_DIR" to a directory containing one of the above
  files.  If "hardware_interface" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  ros_canopen/canopen_motor_node/CMakeLists.txt:7 (find_package)

ipa-mdl, 2015-09-04:
`hardware_interface` is a catin package from ros_control (https://github.com/ros-controls/ros_control/tree/indigo-devel/hardware_interface).
Please install all missing dependencies via `rosdep install`.

droter, 2015-09-09:
Thank you I installed the hardware_interface and the error went away.

closed 2015-09-09 by droter