ros-industrial/ros_canopen #129: fatal error: std_srvs/Trigger.h

droter, 2015-09-09:
Hi,

I am getting this error when trying to compile under indigo on ubuntu 14.04 LTS.

-- +++ processing catkin package: 'canopen_motor_node'
-- ==> add_subdirectory(ros_canopen/canopen_motor_node)
-- checking for module 'muparser'
--   package 'muparser' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal)
  ros_canopen/canopen_motor_node/CMakeLists.txt:22 (pkg_check_modules)

Any ideas?

Thanks for your help,

Matt

droter, 2015-09-09:
Installed std_srvs from here http://wiki.ros.org/std_srvs

closed 2015-09-09 by droter

ipa-mdl, 2015-09-09:
again: please use rosdep, it will install all dependencies:
````
rosdep install --iy --from-paths YOUR_PATH_TO_ROS_CANOPEN
```