ros-industrial/ros_canopen #130: canopen_ch

droter, 2015-09-09:
Hi,

I am getting this error when trying to compile in indigo on ubuntu 14.04 LTS.

/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp: In member function ‘bool canopen::RosChain::setup_sync()’:
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: error: no matching function for call to ‘ros::NodeHandle::param(const char [11], int)’
         if(sync_nh.param("silence_us", 0) != 0){
                                         ^
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: note: candidate is:
In file included from /opt/ros/indigo/include/ros/ros.h:45:0,
                 from /home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/include/canopen_chain_node/chain_ros.h:7,
                 from /home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:2:
/opt/ros/indigo/include/ros/node_handle.h:1912:8: note: template<class T> void ros::NodeHandle::param(const string&, T&, const T&) const
   void param(const std::string& param_name, T& param_val, const T& default_val) const
        ^
/opt/ros/indigo/include/ros/node_handle.h:1912:8: note:   template argument deduction/substitution failed:
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: note:   candidate expects 3 arguments, 2 provided
         if(sync_nh.param("silence_us", 0) != 0){
                                         ^
make[2]: *** [ros_canopen/canopen_chain_node/CMakeFiles/canopen_ros_chain.dir/src/ros_chain.cpp.o] Error 1
make[1]: *** [ros_canopen/canopen_chain_node/CMakeFiles/canopen_ros_chain.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

Thanks for your help,

Matt

ipa-mdl, 2015-09-09:
I was not able to reproduce this error with my test VM (trusty, amd64, g++-4.8.8, indigo), the jenkins devel job succeeded as well.
The code looks fine, the template parameter should get deduced to `int`.  

What system do you use?

ipa-mdl, 2015-09-12:
Does is work now with you new setup?

ipa-mdl, 2015-09-18:
No response and not reproducible.. I will close it for now.
Feel free to reopen it if you still cannot build the code.

closed 2015-09-18 by ipa-mdl