ros-industrial/ros_canopen #131: Installing Dependencies

droter, 2015-09-10:
Hi,

I ran this command rosdep install -i -y --from-paths ~/catkin_ws/src/ros_canopen/  and am getting this message.  I have installed std_msgs but it still isn't able to find it.  Is there something else I need to do?

canopen_chain_node: Cannot locate rosdep definition for [std_msgs]
canopen_master: Cannot locate rosdep definition for [class_loader]
canopen_402: Cannot locate rosdep definition for [rosunit]
canopen_test_utils: Cannot locate rosdep definition for [roslaunch]
canopen_motor_node: Cannot locate rosdep definition for [urdf]

Thanks for your help,

Matt

ipa-mdl, 2015-09-11:
Have you sourced your setup.bash beforehand?

droter, 2015-09-11:
Hi,

Yes, I have sourced the setup file and restarted my machine but I am getting the same results.

Matt

gavanderhoorn, 2015-09-11:
> Yes, I have sourced the setup file and restarted my machine but I am getting the same results.

In that order? Because you will have to source `setup.bash` every time you want/need to run ROS binaries (unless you've added the `source ..` step to your `~/.bashrc`.

droter, 2015-09-11:
Right I restarted and then sourced setup.bash.

droter, 2015-09-11:
This is the error message when I run catkin_make:
catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp: In member function ‘bool canopen::RosChain::setup_sync()’:
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: error: no matching function for call to ‘ros::NodeHandle::param(const char [11], int)’
         if(sync_nh.param("silence_us", 0) != 0){
                                         ^
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: note: candidate is:
In file included from /opt/ros/indigo/include/ros/ros.h:45:0,
                 from /home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/include/canopen_chain_node/chain_ros.h:7,
                 from /home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:2:
/opt/ros/indigo/include/ros/node_handle.h:1912:8: note: template<class T> void ros::NodeHandle::param(const string&, T&, const T&) const
   void param(const std::string& param_name, T& param_val, const T& default_val) const
        ^
/opt/ros/indigo/include/ros/node_handle.h:1912:8: note:   template argument deduction/substitution failed:
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: note:   candidate expects 3 arguments, 2 provided
         if(sync_nh.param("silence_us", 0) != 0){

ipa-mdl, 2015-09-11:
Do not cross-post, please.
You have already filed #130 for this.

Does rosdep succeeds if you source setup.bash after restart, but before rosdep?

droter, 2015-09-11:
Sorry for the cross post.

This is what I am getting after a restart.

viki@c3po:~/catkin_ws$ sudo rosdep install -i -y --from-paths ~/catkin_ws/src/ros_canopen/
[sudo] password for viki: 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
socketcan_interface: Cannot locate rosdep definition for [rosunit]
canopen_chain_node: Cannot locate rosdep definition for [std_msgs]
canopen_master: Cannot locate rosdep definition for [class_loader]
canopen_402: Cannot locate rosdep definition for [rosunit]
canopen_test_utils: Cannot locate rosdep definition for [roslaunch]
canopen_motor_node: Cannot locate rosdep definition for [urdf]
viki@c3po:~/catkin_ws$ source devel/setup.bash 
viki@c3po:~/catkin_ws$ sudo rosdep install -i -y --from-paths ~/catkin_ws/src/ros_canopen/
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
socketcan_interface: Cannot locate rosdep definition for [rosunit]
canopen_chain_node: Cannot locate rosdep definition for [std_msgs]
canopen_master: Cannot locate rosdep definition for [class_loader]
canopen_402: Cannot locate rosdep definition for [rosunit]
canopen_test_utils: Cannot locate rosdep definition for [roslaunch]
canopen_motor_node: Cannot locate rosdep definition for [urdf]

gavanderhoorn, 2015-09-11:
On 11-9-2015 20:15, droter wrote:
> Sorry for the cross post.
> 
> This is what I am getting after a restart.
> 
> viki@c3po:~/catkin_ws$ sudo rosdep install -i -y --from-paths ~/catkin_ws/src/ros_canopen/
> [sudo] password for viki: 

Don't run rosdep with sudo.

sudo doesn't (by default) re-use the environment of the user that
started it, so no ROS environment variables are present for rosdep to use.

rosdep should always be invoked as a normal user, and it will ask for
your password whenever it needs to do something that requires elevated
privileges.

droter, 2015-09-11:
Hi Gavan,

Yes, it was able to run without sudo.  I got this message: #All required rosdeps installed successfully

When I tried to run catkin_make I get the build error from before:

/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp: In member function ‘bool canopen::RosChain::setup_sync()’:
/home/viki/catkin_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp:275:41: error: no matching function for call to ‘ros::NodeHandle::param(const char [11], int)’
         if(sync_nh.param("silence_us", 0) != 0){
...

Thanks for your help,

Matt

ipa-mdl, 2015-09-11:
Which setup.bash have you sourced?
I had some issues with rosdep if I did not source the setup.bash from my catkin workspace, but just /opt/ros/indigo/setup.bash.
It always reported  "All required rosdeps installed successfully" even if this was not right.
Does `/usr/include/muParser.h` exist in your system? (this checks the muparser <depend>).

Can you try to reproduce the problems in a fresh catkin workspace?
Perhaps it is enough to (re)move the build and devel directories (**source devel/setup.bash beforehand!**)

droter, 2015-09-12:
Hi,

I installed a new version of ros and tried it again.  Everything worked correctly.  

There must be a problem with my install.

Thank you for your time,

Matt

closed 2015-09-12 by droter