ros-industrial/ros_canopen #133: 'CAN not ready' after halt/recover

mistoll, 2015-09-18:
When calling the recover service (after halt) the service responds:
> Could not set transition; Could not enable motor

Candump and Node output is here https://gist.github.com/mistoll/6fe24a63e996175d4335

Interesting part of output:
> illegal tranistion 2 -> 6
> [ERROR] [1442578258.554814164]: CAN not ready; Could not quick stop
> illegal tranistion 2 -> 6

mistoll, 2015-09-18:
Maybe fixed by #134

ipa-mdl, 2015-09-18:
I was able to reproduce this behaviour, was introduced in https://github.com/ipa-mdl/ros_canopen/commit/504b4b3736cdc1bd831a025faa9821df3737b570

ipa-mdl, 2015-09-18:
#135 should resolve this issue.
There is still a (very small) chance that recover fails due to bad timing, but the driver should be able to recover from this situation.

mistoll, 2015-09-28:
Resolved with #135
Tested this for a while and didn't happen again.

closed 2015-09-28 by mistoll