ros-industrial/ros_canopen #137: candump to topic  Feature Request

droter, 2015-09-21:
Hi,

Can ros_canopen publish the candump to a topic?

Thanks,

Matt

ipa-mdl, 2015-09-21:
No, this is currently not supported.
Could you give some details on your use case, please?

droter, 2015-09-21:
I would like to record the CAN traffic to a bag file.  I am trying to compare sensor data with vehicle CAN data.  Is there a different approach that would be better?

For the start the raw candump would work.  It would be nice to publish decoded CAN traffic to a topic.

Thanks,

Matt

ipa-mdl, 2015-09-21:
I would just use candump (from can-util) since this is the only tool that read the correct timestamps.
I addition you can replay the message.
Why do you need a rosbag for it?

`canopen_chain_node` can publish simple CANopen objects as topic.

If you need do decode custom CAN messages, then you have to decode them yourself.
A CAN-to-ROS implementation requires 10 lines of code (http://answers.ros.org/question/213645/running-can-radar-on-ros-through-raspberry-pi-2/?answer=215226#post-id-215226).
You can as well filter by ID and decode the messages as needed.

droter, 2015-09-22:
Thank you for the suggestions.

I was looking at a way to get the candump data publish on a topic so I could record it with the sensor data.  Then all of the data would have the same timestamp and I could play it back.  Can the canopen_chain_node get the candump data?

Matt

droter, 2015-09-22:
The link to the ros answers is great.  Missed that in searching.  I think CAN for ros is a good thing to work on and appreciate all you guys are doing.

ipa-mdl, 2015-09-23:
If you really want to publish CAN raw (can header + payload data) frames then you could inherit from `canopen_chain_node::RosChain` and add your publisher.
But `canopen_chain_node` is not meant to handle CAN frames, it is meant for CANopen networks.

`ros_canopen` does not cover non-CANopen devices, perhaps the CAN package might be moved into a new repo anyway (#103).

closed 2015-10-04 by droter

iwanders, 2015-12-23:
Hi,

I'm working with a non-CANopen network and would like to have all CAN messages available a topic as well. This would allow recording of all messages sent over the bus in a bagfile and allow other ROS nodes to convert the messages on the CAN topic into the appropriate types and topics.

Yesterday I spent some time to get to a working proof of concept, it uses the `ThreadedSocketCANInterface` with callbacks for the messages. Since I've seen multiple posts about such a node I'd like to send a pull request back for this node. Before continuing I'm soliciting some feedback.

* Would you accept a pullrequest for this type of node? Is it acceptable to make it a part of the `socketcan_interface` package and name it `can_to_topic`?, that seems to be a reasonable location because the node which just puts all CAN frames on a topic is heavily dependent on this package?
* As far as I know, there currently is no CAN message type in ROS (if there was, there probably would be a node producing it as well...) I would propose naming it `can_frame` with the following fields:
  * Header header
  * uint32 id
  * bool is_rtr
  * bool is_extended
  * uint8 dlc
  * uint8 data[8]
  
  The is_error boolean can be left out I presume? As the comment next to this attribute in the Header definition states it is only for internal use? Perhaps it would be useful to send a `ROS_WARNING` message out for those? Judging from the wiki pages `Header` seems to be preferred over `time`, even though the `frame_id` is not really needed in our case.

* I would propose the Topic name to publish on to be something like `/can/incoming`. Future development would allow the node to subscribe to `/can/outgoing` and publish those messages back on the bus.

Any other comments / tips / remarks are very welcome.

ipa-mdl, 2016-01-07:
Hi :)

As pointed out above I am not that keen on a CAN-to-ROS relay, but I see that there is a demand for this feature.
So a pull request is more than welcome.

Some remarks:

> Is it acceptable to make it a part of the `socketcan_interface` package 

No it's not.This package does not have any other ROS dependencies (`class_loader` is ROS-independent). Please create a new package for it.
It would be great if it could be reused in other ROS nodes as well, so a clean code separation in terms of classes would be great.

> As far as I know, there currently is no CAN message type in ROS (if there was, there probably would be a node producing it as well...)

https://github.com/FroboLab/frobomind/tree/master/fmLib/computer_support/can_socketcan
and https://github.com/juancamilog/canbus_interface do provide the functionality and some msg types.
None of them was released, so I guess it makes sense to create a new message type.
For the sake of reuse it should be in a separate msg package (`can_msgs` ?).

Your msg proposal looks fine, but I would add the error flag as well, since it is part of the CAN frame.

> I would propose the Topic name to publish on to be something like `/can/incoming`

Please do not use absolute names.I think `incoming` is ambiguous, how do you like `received_messages`?

iwanders, 2016-01-08:
I made a start with this today, I've spent the major part of the day reading on the ROS wiki and looking at how this and other packages were structured. I learned a lot by doing so; for one, I now understand that it would be a no-go to add the functionality to the `socketcan_interface` package.

I've introduced the `can_msgs` packages you proposed in cf17e15ffa36d63a1f3d9ce154ca93979ba6c187, I ultimately named the frame `can_msgs/Frame`, this matches the other message naming conventions. The entries of the `Frame` are identical to attributes of the `can::Frame` from `socketcan_interface`.

For the package that provides the actual CAN to Topic functionality, I chose `socketcan_bridge` as name, there are quite a few other packages which use the `_bridge` suffix to denote that it deals with data going to and from ROS.

The implementation I worked on today can be seen in 2a968f3cc405276519f905923ea62b111b593407.  I was not entirely sure what you meant by:
>It would be great if it could be reused in other ROS nodes as well, so a clean code separation in terms of classes would be great.

I tried searching for some other packages where a class is reused, but I could not find any. Perhaps I already used such a structure: To me it made sense to create a `socketcan_to_topic` class (currently requires `ThreadedSocketCANInterface` as argument), the intention was that the counterpart `topic_to_socketcan` class could share the same SocketCAN object. By structuring the class in such a way it could be reused in a uni-directional node but also in a bidirectional bridge, is this what you meant?

However, as you probably guessed I hit a snag with the `ThreadedSocketCANInterface`; it does not expose the `send()` method from the `SocketCANInterface`. Earlier in this thread you proposed to subclass from `canopen_chain_node::RosChain` and add your own publisher, but I think it this would introduce more complexity than is desired. I suppose it is better to use the `SocketCANInterface` instead of the already threaded one?

iwanders, 2016-01-10:
I realized that I misinterpreted the header files last Friday, so despite my earlier statement the `send()` method is actually available from the threaded interface. So the implementation to go from ROS messages to SocketCAN was pretty straightforward.

In commit c1532c11ff53342f5825c295dcfb3d35cbad858a; I introduced the `TopicToSocketCAN` class, its node and a node which combines the two classes to provide a bidirectional bridge. The newest commit, 489ef5453b3b876ad78b4303e0a990d1271104ea, adds a testcase in which the conversion between the `can_msgs::Frame` message and `can::Frame` struct is checked.

I'm not entirely sure how to include testing the classes in the default testing framework, currently I've tested them using the `vcan0_to_vcan1.launch` file and verifying the output of `candump vcan0 vcan1`. One option I can think of is to modify your `DummyInterface` class such that it inherits from `DriverInterface` instead of only from `CommInterface`, then that dummy interface can be used in loopback mode to simulate incoming messages from SocketCAN.

Another addition that's perhaps useful is to provide some way to use the masking functionality offered by `createMsgListener`, this might be useful if you are only interested in a small portion of the CAN messages, but perhaps that's something for a future enhancement.

I look forward to hearing your thoughts on the progress so far.

ipa-mdl, 2016-01-19:
I just had a brief look, but your code looks great :)
Feel free to extend `DummyInterface`, it is meant just for unit tests.

Please open a PR if you are ready, I will review it as soon as I can.

> Another addition that's perhaps useful is to provide some way to use the masking functionality offered by createMsgListener, this might be useful if you are only interested in a small portion of the CAN messages, but perhaps that's something for a future enhancement.

It could even read a config and publish as many topics as needed with their respective masks.
The `can::toheader` might come in handy for this.