ros-industrial/ros_canopen #141: Limit interface do not play well with ros_canopen

ipa-mdl, 2015-09-30:
The current implementation of https://github.com/ros-controls/ros_control/tree/indigo-devel/joint_limits_interface does not work properly for all use cases, see https://github.com/ros-controls/ros_control/issues/193 as well.

ipa-mdl, 2015-10-01:
* plug-based limiter (output filter) system
* implementation of open-loop limiter
* end position limiter: full stop
* per controller configuration

ipa-mdl, 2015-11-28:
* remove joint_limit_interface code in favour of https://github.com/ipa320/cob_control/issues/72
* restrict 402 commands to bounds according to mode
* URDF will not be needed anymore (addresses #145)

ipa-mdl, 2016-02-22:
* warn about missing limiter (indigo only)

ipa-mdl, 2016-05-19:
Preliminary implementation is done in https://github.com/ros-industrial/ros_canopen/tree/limited_joint_handle

ipa-fxm, 2017-01-21:
A summary about current approaches and insights regarding "limit interfaces" is given in https://github.com/ipa320/cob_gazebo_plugins/issues/14#issuecomment-274255606

ipa-mdl, 2017-01-21:
For now `joint_limits_interface` can be turned off for specific controllers (#215 and #216).