ros-industrial/ros_canopen #149: Exception during parsing of eds

ipa-fxm, 2015-10-23:
When trying to run canopen_motor_node like this (to be used with an Danaher ACS-XS 6 MotorController)
```
<?xml version="1.0"?>
<launch>

  <include file="$(find fieldmobile_base)/launch/upload_robot.launch"/>

    <!-- robot state publisher -->
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
        <param name="publish_frequency" type="double" value="50.0" />
        <param name="tf_prefix" type="string" value="" />
    </node> 

  <!-- ros_canopen driver -->
  <node name="driver" pkg="canopen_motor_node" type="canopen_motor_node" output="screen" clear_params="true" launch-prefix="">
    <rosparam command="load" file="$(find fieldmobile_bringup)/config/can0.yaml" />
    <rosparam command="load" file="$(find fieldmobile_bringup)/config/canopen_driver.yaml" />
  </node>

  <!--include file="$(find fieldmobile_control)/launch/fieldmobile_control.launch"/-->

</launch>
```

We receive the following exception:
```
$ roslaunch fieldmobile_bringup fieldmobile_canopen.launch 
... logging to /home/build/.ros/log/42da1dd0-797b-11e5-8cab-f816548b8a08/roslaunch-pokini07-23718.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pokini07:37766/

SUMMARY
========

CLEAR PARAMETERS
 * /driver/

PARAMETERS
 * /driver/bus/device: can0
 * /driver/defaults/eds_file: config/69G32135C0...
 * /driver/defaults/eds_pkg: fieldmobile_bringup
 * /driver/name: fieldmobile
 * /driver/nodes/wheel_rl_joint/id: 1
 * /driver/nodes/wheel_rr_joint/id: 2
 * /driver/sync/interval_ms: 10
 * /driver/sync/overflow: 0
 * /manipulator/robot_state_publisher/publish_frequency: 50.0
 * /manipulator/robot_state_publisher/tf_prefix: 
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /use_sim_time: True

NODES
  /manipulator/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
  /
    driver (canopen_motor_node/canopen_motor_node)

auto-starting new master
process[master]: started with pid [23733]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 42da1dd0-797b-11e5-8cab-f816548b8a08
process[rosout-1]: started with pid [23746]
started core service [/rosout]
process[manipulator/robot_state_publisher-2]: started with pid [23771]
process[driver-3]: started with pid [23803]
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::property_tree::ptree_bad_path> >'
  what():  No such node (SubNumber)
[driver-3] process has died [pid 23803, exit code -6, cmd /home/build/catkin_ws/devel/lib/canopen_motor_node/canopen_motor_node __name:=driver __log:=/home/build/.ros/log/42da1dd0-797b-11e5-8cab-f816548b8a08/driver-3.log].
log file: /home/build/.ros/log/42da1dd0-797b-11e5-8cab-f816548b8a08/driver-3*.log
```

More information on the configuration can be found in this [gist](https://gist.github.com/ipa-fxm/9178a5c1558acfd7e77e)

ipa-fxm, 2015-10-23:
Using another eds file with only the mandatory objects, parsing starts but then complains:
```
$ roslaunch fieldmobile_bringup fieldmobile_canopen.launch 
... logging to /home/build/.ros/log/14d0525c-797f-11e5-8bad-f816548b8a08/roslaunch-pokini07-25667.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pokini07:39056/

SUMMARY
========

CLEAR PARAMETERS
 * /driver/

PARAMETERS
 * /driver/bus/device: can0
 * /driver/defaults/blub: 0
 * /driver/defaults/eds_file: config/exp.eds
 * /driver/defaults/eds_pkg: fieldmobile_driver
 * /driver/name: fieldmobile
 * /driver/nodes/wheel_rl_joint/id: 1
 * /driver/nodes/wheel_rr_joint/id: 2
 * /driver/sync/interval_ms: 10
 * /driver/sync/overflow: 0
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /robot_state_publisher/tf_prefix: 
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    driver (canopen_motor_node/canopen_motor_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [25682]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 14d0525c-797f-11e5-8bad-f816548b8a08
process[rosout-1]: started with pid [25695]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [25713]
process[driver-3]: started with pid [25738]
[ERROR] [1445602243.470579437]: /home/build/catkin_ws/src/ros_canopen/canopen_master/include/canopen_master/objdict.h(213): Throw in function const boost::shared_ptr<const canopen::ObjectDict::Entry>& canopen::ObjectDict::at(const canopen::ObjectDict::Key&) const
Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<std::out_of_range> >
std::exception::what: Unable to find 6041 in dictionary

[driver-3] process has died [pid 25738, exit code 255, cmd /home/build/catkin_ws/devel/lib/canopen_motor_node/canopen_motor_node __name:=driver __log:=/home/build/.ros/log/14d0525c-797f-11e5-8bad-f816548b8a08/driver-3.log].
log file: /home/build/.ros/log/14d0525c-797f-11e5-8bad-f816548b8a08/driver-3*.log
```

In general, we do not have any object in the 6k range....
Does ros_canopen rely on finding these objects?
What part of the eds is actually used/mandatory at all as every object can be overwritten using dcf_default and id-specific over-writes

ipa-mdl, 2015-10-23:
Your EDS file does not conform to the standards in many ways..
I cannot test it now, will do that later.

> In general, we do not have any object in the 6k range....

`canopen_motor_node` can be used for CANopen 402 devices only. This device profile uses only 0x6000 to 0x67FF objects; this is the reserved namespace for profile specific-data.
It can use up to 7 additional blocks (0x6800 to 0x6FFF,  0x7000 to 0x77FF) for logical devices, but this not supported by the current  `canopen_402` implementation.

> Does ros_canopen rely on finding these objects?

The packages rely on mandatory objects (according to the CANopen standards, i.e. 301 and 402).
The different layers use optional objects as well, but have fallbacks for all(?) of them.

> What part of the eds is actually used/mandatory at all as every object can be overwritten using dcf_default and id-specific over-writes

* `canopen_master` uses objects form the 1000-1FFF communication area, basically SDO, PDO, and heartbeat settings. Every CANopen device has to comply with the 301 standard, which defines these objects.

* `canopen_chain_node` can set dcf_overlays and publish objects. The listed objects have to be included in the EDS. `canopen_master` objects are need as well.

* `canopen_402` objects are listed here: http://wiki.ros.org/canopen_402

* `canopen_chain_node`: `canopen_master` +  `canopen_chain_node`  +  `canopen_402`  + objects listed on *_from_device settings (per default 6064 and 606C).

ipa-mdl, 2015-10-23:
Apparently your controller does not support the CANopen 402 profile.
You can use `canopen_motor_node` if you provide a custom motor protocol plugin (https://github.com/ros-industrial/ros_canopen/blob/indigo-devel/canopen_402/include/canopen_402/base.h).

ipa-mdl, 2015-10-26:
The exception was thrown because the sub objects `1a0Xsub1` have objects type 8 (=Array) and the EDS parser read them recursively.  Object type 8 is not valid here, only object type 7 (=Variable) should be used for sub-objects.
For the sake of convenience I have decided to relax the parsing and assume object type 7 for all sub objects (3ce4c9cec24a59ab93b857e68dbc3eddc74161b5).

So it should not crash anymore, but it will not work anyway as I pointed out before.

ipa-fxm, 2015-10-26:
Thanks, we were already assuming that the protocol is not 402-conform...thanks for confirming that...
We will discuss your pointer to implement the plugin for our device!
Until then, we can close this issue.

closed 2015-10-26 by ipa-fxm