ros-industrial/ros_canopen #160: socketcan_bridge support for DBC files

reinzor, 2016-02-13:
Add encoding/decoding step to the socketcan_bridge:

- Load DBC Files that contain Frame/Signal specification (http://vector.com/vi_candb_en.html)
- Create pub and sub for DecodedCANFrames

ipa-mdl, 2016-02-17:
Do you plan to work on this or is it just a feature request?

reinzor, 2016-02-17:
This is a feature request. If somebody is interested, I can help developing
this feature.

-Rein

iwanders, 2016-02-17:
I don't really see how you envision the `DecodedCANFrames` message to work or what it benefits will be. Of course, for some CAN messages such a conversion node would be useful for example CAN enabled encoders could be conveniently converted to `sensor_msgs/JointState`. I guess you could make a node which converts specific CAN messages to specific ROS messages. Should be an additional package in my opinion though.

ipa-mdl, 2016-02-17:
I have done something similar for `canopen_chain_node`: each object can be published as a single topic with one of the `std_msgs` types.

reinzor, 2016-02-17:
@ipa-mdl,This indeed could be an option; however then you loose the relation between signals of the same CAN Frame. @iwanders, you might be right, maybe a more nice way is developing an abstract base class with a decodedframe callback that maps the decoded can frames on ROS messages ...

ipa-mdl, 2016-02-18:
If you really want to publisher complex messages then you should go for a python-can implementation or a converter node that listens to the socketcan_bridge output.

melvinpmanuel, 2022-08-09:
> 

Hello Mike,
Suppose I have a .dbc file and if I need to visualize the data in a human readable format, what we shall do? Do we have some documentation for proceeding further?