ros-industrial/ros_canopen #161: Jade/Kinetic adpations

ipa-mdl, 2016-02-22:
Mode switching needs to support mixed joint settings.

ipa-mig, 2016-06-23:
FYI, there has been a C++ API breaking change in ros control: https://github.com/ros-controls/ros_control/commit/afaf9403d1daf6e7f0a93e4a06aa9695e2883632

ipa-mdl, 2016-06-23:
That's what I meant with  `mixed joint settings` ;)

ipa-mig, 2016-06-23:
Ooops :-D

mistoll, 2016-11-17:
I'd like to provide a pr for this.
It could be solved with a simple additional for loop over claimed_resources. Are there any doubts about that?
How do you wan't to select between the old and the new api? (Separate branch, conditional compilation)

ipa-mdl, 2016-11-18:
> How do you want to select between the old and the new api?

separate branch, `jade-devel` for jade and kinetic.

> It could be solved with a simple additional for loop over claimed_resources. Are there any doubts about that?

This works only for controllers without mixed modes.
However, it should be okay to just loop over the claimed_resources, but fail if more than one resource type is needed. (-> this supports all indigo-based controllers)

Mixed mode settings should be supported as well, but can be added later.

PR #198 (2016-11-26): Jade fixes
supersedes #196, fixes #161

@mistoll: FYI
PR #200 (2016-12-13): Support mixed modes
closes #197, closes #161
closed 2016-12-13 by ipa-mdl