ros-industrial/ros_canopen #162: License change

ipa-mdl, 2016-02-23:
Candidates:
* Mozilla Public License 2.0
* BSD
* Apache License 2.0

iwanders, 2016-02-23:
What's the motivation for proposing a license change?

I'm not a lawyer so I'm somewhat hesitant to comment on these matters, but I believe changing a license requires each individual contributor to agree to the proposed license change or their edits have to be removed. This is what the [OpenStreetMap][openstreetmaplicense] project did when they decided to change the license. The number of contributors is relatively low, so even if this is indeed required it should be doable.

For what it's worth; I'd say BSD, it is the [recommended license][developerguide] to be used for ROS ('Reasons for choosing BSD' on that page provides some arguments). They strongly recommend BSD for `.msg` and `.srv` files, so I chose that for the `can_msgs` package. Since `socketcan_interface` is LGPL and the `socketcan_bridge` only links to this LGPL licensed part I believe it was acceptable to license it as BSD in accordance to the ROS recommendation.

[developerguide]: http://wiki.ros.org/DevelopersGuide#Licensing
[openstreetmaplicense]: http://wiki.openstreetmap.org/wiki/Open_Database_License

gavanderhoorn, 2016-02-23:
@iwanders wrote:
> For what it's worth; I'd say BSD, it is the [recommended license][developerguide] to be used for ROS ('Reasons for choosing BSD' on that page provides some arguments).

For ROS 1 that is certainly true. For ROS 2 the OSRF is mostly using Apache 2 (because of some improved protection wrt patent claims and the included CLA, among other things).

In ROS-Industrial we're also using Apache 2 for new developments.

iwanders, 2016-02-23:
> because of some improved protection wrt patent claims and the included CLA, among other things
In ROS-Industrial we're also using Apache 2 for new developments.

The Apache 2 license is indeed a lot more explicit in regard to the patents and Contributions. If using the Apache License 2.0 can prevent patent problems and the like I don't have any objections to using this over the BSD license I chose initially.

ipa-mdl, 2016-02-23:
> changing a license requires each individual contributor to agree to the proposed license

Yes you're right, but most contributors are Fraunhofer IPA employees. `can_msgs` and `socketcan_bridge` are already BSD licensed. You might decide to change as well, but you don't have to.

This issue is not meant for discussion, it is just a reminder for the transition .

gavanderhoorn, 2016-02-23:
> > changing a license requires each individual contributor to agree to the proposed license

> Yes you're right, but most contributors are Fraunhofer IPA employees. `can_msgs` and `socketcan_bridge` are already BSD licensed. You might decide to change as well, but you don't have to.

But afaik @iwanders is right: you can't *change* the license of a project without consent from all developers (you could fork everything and continue under a different license).

Unless of course you mean that because `can_msgs` and `socketcan_bridge` are separate packages it doesn't matter anyway.

iwanders, 2016-02-23:
> (you could fork everything and continue under a different license).

That is only possible if the two licenses are compatible, but that is probably what you meant.

> Unless of course you mean that because `can_msgs` and `socketcan_bridge` are separate packages it doesn't matter anyway.

Yep, if the license needs to be changed for these packages it is easier since there are few contributors.

> This issue is not meant for discussion, it is just a reminder for the transition .

I see, when you have settled on one let me know. I think it would be nice if all packages have the same license. (Currently the `ros_canopen` metapackage has an LGPL license, but it also installs the `socketcan_bridge` which is BSD, not a big deal, but it might be confusing.)

ipa-mdl, 2016-02-23:
I have clarified in the description that this issue covers the LGPL code only.

> you can't change the license of a project without consent from all developers

For employees all rights belong to the employer (within the scope of the employment). Even I have no rights here.

> Currently the ros_canopen metapackage has an LGPL license, but it also installs the socketcan_bridge which is BSD

I suppose the metapackage's license covers the metapackage's code (package.xml and CMakeLists.txt) only, not the packages it depends on.

ipa-mirb, 2016-05-18:
Hi all, any news regarding this issue?

ipa-mdl, 2016-05-19:
We will switch to BSD licence with the next release.

BenArtes, 2019-09-13:
What is the status of this? It seems as though the license wasn't updated, are there plans to?

Asking as I'm starting a project that has licensing restrictions preventing us from maintaining LGPL'd code and we're likely to have to make modifications to ros_canopen to support our project.

Some of the changes are likely to be of use to the wider community (Homing Improvements etc) but others are custom details and features that wouldn't have a use outside of our Company / application that we don't wish to open source.

If the license stays LGPL I'll be unfortunately forced (by company policy) to go ahead with a custom implementation and not able to work on features that I think would improve some areas of ros_canopen.

ipa-mdl, 2019-09-14:
I don't think this will change any time soon.
I have considered migrating to MPL 2.0, which only has a weak copyleft, but it does not seem to be worth the effort.

`ros_canopen` offers many ways of extending the functionionality without changing the code (subclassing, plug-ins), and we can expose more customization interfaces if needed.

gavanderhoorn, 2019-09-14:
@ipa-mdl: what changed between https://github.com/ros-industrial/ros_canopen/issues/162#issuecomment-220273416 and https://github.com/ros-industrial/ros_canopen/issues/162#issuecomment-531481441?

ipa-mdl, 2019-09-18:
> what changed between #162 (comment) and #162 (comment)?

Change of priorities..

gavanderhoorn, 2019-09-19:
That's a bit vague? Priorities changed so you had to work on other things? Or priorities wrt potential "markets" for `ros_canopen`?

dineshr93, 2019-09-25:
Can I take can_msgs of Melodic branch is in BSD? Because it is causing a problem if the license is LGPL if not Dynamically liked with proprietary source.

ipa-mdl, 2019-09-25:
> Can I take can_msgs of Melodic branch as BSD? (@dineshr93)

Yes. `can_msgs` and `socketcan_bridge` are BSD licensed, as per decision of the author.

> That's a bit vague? Priorities changed so you had to work on other things? (@gavanderhoorn) 

My spare time is limited and changing the license is not on my list (anymore).
To reflect this change, I will close this issue.

closed 2019-09-25 by ipa-mdl