ros-industrial/ros_canopen #188: Add Timestamping to socketcan_interface

Worldman2, 2016-08-10:
There should be a way to obtain the timestamp of receiption for a CAN-frame. Ideally, this should be the HW-timestamp, if the device supports it (using SIOCSHWTSTAMP). According to the socketcan documentation, timestamps can be obtained by an ioctl-call after reading a message from the socket.

I tried hacking something into socketcan.h but failed.
My approach was to just add 

    ioctl(socket_.native_handle(), SIOCGSTAMP, &tv);

at the beginning of SocketCANInterface::readFrame(...) and 

    setsockopt(sc, SOL_SOCKET, SO_TIMESTAMP, &timestamp_on, sizeof(timestamp_on));

in the init(...) method.
However, it only seems to work after receiving the third-ish message, so I do not trust it and it probably just proves how few I understand about socketcan and boost::asio. 

Is there a possibility for a recent implementation of this feature request or could you provide some hints how to realize this? Thanks!

ipa-mdl, 2016-08-10:
I tried it myself some time ago, but for some reason I did not include it somewhere.
If I remember it correctly, not all devices set these time stamps and in addition it is unclear what time base they use.

However, timestamps would be a nice feature!

Worldman2, 2016-08-10:
As far as I understand, even if the hardware does not support it, there is at least the option to retrieve the socket timestamp (SO_TIMESTAMP) of the last received packet. I think, this is also what candump does: https://github.com/linux-can/can-utils/blob/master/candump.c#L676

ipa-mdl, 2016-12-21:
Found my implementation again: https://github.com/ipa-mdl/ros_canopen/tree/stamped_data