ros-industrial/ros_canopen #189: Indigo release

mistoll, 2016-08-29:
Are you going to release an up to date version of ros_canopen to indigo?

Pro:
- Most issues are enhancements or are targeted for jade/kinetic. (Only #184 is relevant to indigo).
- There haven't been any commits recently.
- (There's a final_indigo_release milestone)

Con:
- There are breaking changes to the current indigo release 0.6.4.

ipa-fmw, 2016-08-29:
currently there is version 0.6.4 released from 3rd of July 2015. There are definitively worthwhile updates and fixes. I would say "yes" to another indigo release.

@ipa-mdl: what's you opinion?

ipa-mdl, 2016-09-13:
> Are you going to release an up to date version of ros_canopen to indigo?

yes

Open Issues:
- 3 open PRs need to be tested
- Licence change (#162)

ipa-mdl, 2016-12-10:
released: https://github.com/ros/rosdistro/pull/13381

ipa-mdl, 2016-12-11:
and now in testing :)

closed 2016-12-11 by ipa-mdl