ros-industrial/ros_canopen #190: SDO reading timeout

Chickenman222, 2016-09-13:
Hi

I am using schunk_lwa4p and you ros_canopen package. I have problem with reading temperatures from lwa4p. I added this line 
`publish: ["200Dsub1!"]`
to arm_driver.yaml file which created 6 topics with temperature of each joint. But when I create trajectory loop for my arm I am getting this error randomly:

> [ERROR] [1473778918.693696206]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 2.129656 seconds). Stopping trajectory.
> [ INFO] [1473778918.693760171]: MoveitSimpleControllerManager: Cancelling execution for /arm/joint_trajectory_controller
> [ INFO] [1473778918.693855586]: Execution completed: TIMED_OUT
> Did not receive a response message
> Could not process message
> [ERROR] [1473778918.823180530]: /home/lwa4p/catkin_ws/src/ros_canopen/canopen_master/src/sdo.cpp(428): Throw in function void canopen::SDOClient::transmitAndWait(const canopen::ObjectDict::Entry&, const canopen::String&, canopen::String*)
> Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException> >
> std::exception::what: SDO
> [canopen::tag_objectdict_key*] = 200dsub1
> 
> Could not process message
> discarded message 
>

This stops trajectory execution and I need to restart it. This problem also appears when I am reading  efforts with this command `eff_from_device: "obj6077"`. Is there a way to avoid this issue ? 
In my opinion it is pointless to read temperature from device in synch interval but there is probably no other way to read it.

Thank you for you answer and also for all the work you do.

ipa-mdl, 2016-09-13:
>  Is there a way to avoid this issue ? 

SDO access is blocking and the publish function reads the data directly (it is meant for debugging).
You should map `200dsub1` and `obj6077` to PDOs to not interrupt the control loop.
(in *.dcf or in dcf_overlay paramter).

I guess you get the control timeouts because the control loop uses the requested cycle rate and not the real timing per default (better control behaviour for the "slow" update rate).
And since your SDO access is blocking, your control loop cannot meet its rate and exceeds the interpolation timeout.

> In my opinion it is pointless to read temperature from device in synch interval but there is probably no other way to read it.

You can reduce the CAN bus load by tweaking the PDO transmission type.
However, the publish rate will still be the control rate.

Chickenman222, 2016-09-13:
Hi 

Thank you for your answer I did not expect so fast response. Can I please ask you how to exactly  map this datas in SDF ?

This is how parameter looks like in my SDF file. 

> 
> [200Dsub1]
> ParameterName=actual_pcb_temperature
> ObjectType=0x7
> DataType=0x0008
> AccessType=ro
> PDOMapping=1

Sorry for asking but I read a lot of responses about maping to PDOs but never  understand how should setup it in sdf file. I was not able to find documentation for that and do not have enough experience with can devices.

ipa-mdl, 2016-09-13:
You can use CANeds (version 3.6 is freely available and works with wine) to alter the PDO mappings in the EDS/DCF.
The `schunk_lwa4p` DCF was patched with the help of https://github.com/ipa320/canopen_test_utils/blob/indigo-devel/scripts/schunk_mapping.py

Chickenman222, 2016-09-21:
Hello

I added object with temperature to the PDO mapping but error is still appearing and it is still the same error. I used CANeds  for editing sdf file but it wont help. Any other ideas ?

Thank you

ipa-mdl, 2016-09-21:
Have you verified (in candump) that really no SDO is sent for the temperature?

ipa-mdl, 2017-03-20:
outdated.
Feel free to reopen the issue if the problem is still around.

closed 2017-03-20 by ipa-mdl