ros-industrial/ros_canopen #191: Build issue with canopen_motor_node robot_layer.cpp

qnetjoe, 2016-10-12:
When I go to build I get the following error:

`/home/jdaily/repo/ros/src/ros_canopen/canopen_motor_node/src/robot_layer.cpp: In member function ‘virtual bool RobotLayer::prepareSwitch(const std::__cxx11::list<hardware_interface::ControllerInfo>&, const std::__cxx11::list<hardware_interface::ControllerInfo>&)’:
/home/jdaily/repo/ros/src/ros_canopen/canopen_motor_node/src/robot_layer.cpp:340:80: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘resources’
             for (std::set<std::string>::const_iterator res_it = controller_it->resources.begin(); res_it != controller_it->resources.end(); ++res_it){
                                                                                ^
/home/jdaily/repo/ros/src/ros_canopen/canopen_motor_node/src/robot_layer.cpp:340:124: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘resources’
             for (std::set<std::string>::const_iterator res_it = controller_it->resources.begin(); res_it != controller_it->resources.end(); ++res_it){
                                                                                                                            ^
ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor_node.dir/build.make:86: recipe for target 'ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor_node.dir/src/robot_layer.cpp.o' failed
`

ipa-mdl, 2016-10-12:
`ros_canopen` is not yet available for `jade` and `kinetic` because of changes in `ros_control`  (#161).

ipa-mdl, 2016-11-18:
@qnetjoe: does #199 work for you?

ipa-mdl, 2016-11-26:
fixed

closed 2016-11-26 by ipa-mdl