ros-industrial/ros_canopen #192: next release?

gavanderhoorn, 2016-10-18:
More of a question, but I was wondering if/when the next release is planned? 0.6.4 is already more than a year old, and there appear to be quite some updates that are only found in the sources at the moment.

gavanderhoorn, 2016-10-18:
I'm aware of #161. I'm specifically asking about Indigo.

ipa-mdl, 2016-10-18:
(duplicate of #189)

I don't have much time for testing right now.
I could release the current state (as it is running on Care-O-bot 4), but without the missing issue.

gavanderhoorn, 2016-10-18:
I hadn't seen #189, my apologies.

I'll be the last to bug other devs about releases, but would there be much difference between users reporting bugs about *from source* builds versus a released build?

victorvorobev, 2016-10-25:
I'm sorry, will there be any release for kinetic? I understand that there is some issues with ros_control, but still?

ipa-mdl, 2016-10-29:
> I'm sorry, will there be any release for kinetic? I understand that there is some issues with ros_control, but still?

A jade/kinetic release can be triggered after https://github.com/ros-industrial/ros_canopen/issues/161.
However, I cannot work on it right now.

ipa-mdl, 2016-12-10:
just released for indigo.
jade will follow after I was able to test it myself.
same for kinetic, but its more complicated to test (still on 14.04).

gavanderhoorn, 2016-12-11:
Very nice. Thanks for the effort.

Would you prefer to track jade/kinetic releases in a separate issue?

ipa-mdl, 2016-12-11:
"Next release" is a nice reminder ;)
The jade and kinetic releases are tracked in https://github.com/ros-industrial/ros_canopen/milestone/4 and https://github.com/ros-industrial/ros_canopen/milestone/5

ipa-mdl, 2016-12-13:
Jade release triggered: https://github.com/ros/rosdistro/pull/13399

ipa-mdl, 2016-12-21:
kinetic is relased as well.

gavanderhoorn, 2016-12-21:
Awesome! :sparkles: :hamburger: :beers:

ipa-mdl, 2017-03-20:
Released again for indigo to kinetic :)

closed 2017-03-20 by ipa-mdl