ros-industrial/ros_canopen #209: Supported drive mode lookup fails with get_cached()

davidmball, 2016-12-23:
I was unable to set a mode for my motor which meant that the hardware_interface::VelocityJointInterface failed to start. 

To fix it in this function
`bool Motor402::isModeSupportedByDevice(uint16_t mode){`

I had to change from get_cached() to get() for the supported_drive_modes_ ie
`    return mode > 0 && supported_drive_modes_.valid() && (supported_drive_modes_.get_cached() & (1<<(mode-1)));`
to
`    return mode > 0 && supported_drive_modes_.valid() && (supported_drive_modes_.get() & (1<<(mode-1)));`

to get it working. After making this change everything works as expect. Is this something I haven't configured correctly?

I am on indigo-devel and using dunkermotors. I am also setting    ` required_drive_mode: 3` in my velocity_controllers/JointVelocityController.

ipa-mdl, 2016-12-23:
`get_cached` should be correct. It checks if the object already has a value, if not it will use `get()`.
The supported drive modes will not change, so this logic is used here.

>  Is this something I haven't configured correctly?

Please check the default of `0x6502` in your EDS/DCF.
It should reflect the correct modes, if not just remove it.

davidmball, 2016-12-23:
Removing the default value from the eds file worked. I can read this value from the motor (it is 0xEF).
I didn't realise that the code would use the value from the eds file as the default cached version. I expected it to still query the motor. 

Thank you for your help and very quick response.

closed 2016-12-23 by davidmball