ros-industrial/ros_canopen #210: Unable to run the test on ros kinetic

mhariharasudan, 2016-12-27:
This is more of a question. I am trying to run the test launch file (hw_rig1.launch under canopen_test_utils) in ros-kinetic and I seem to get te following error after some timeout
"Controller Spawner couldn't find the expected controller_manager ROS interface."

The controller manager does not seem to be running.
Debugging into the code I saw that the ControllerManagerLayer::HandleInit method in control_node.cpp is not being called. Do I have to launch something else in order to trigger the controller manager?

ipa-mdl, 2016-12-29:
Have you called the /rig1/driver/init service?
`canopen_test_utils` is bunch of tools we at IPA use internally and is not really maintained for external use, it was moved into a separate repository (https://github.com/ipa320/canopen_test_utils)

Please do not use `indigo_release_candidate`, it is heavily outdated, and will be removed soon.
(it is still around because it was listed in rosdistro until recently)
`indigo-devel` has the most recent code, `indigo` contains the binary-released code.

ipa-mdl, 2016-12-30:
For ROS kinetic please use `jade-devel` or `jade`.

mhariharasudan, 2016-12-30:
I would like to use this package with my Elmo motors over CAN. Are there any dummy test scripts to understand the program and instructions on configuring it with a new hardware that you can point me towards?

closed 2016-12-30 by ipa-mdl