ros-industrial/ros_canopen #211: Bug in dcf file interpreter?

Linkeway, 2016-12-29:
Hi,

I have successfully used **ros_canopen** to talk to my robot **through position_controller**. However, I failed to use **joint_trajectory_controller**(PVT operation mode) which involves PDO communication.

candump showed that **RPDO2** data on the CAN bus: 

```can0  TX - -  301   [6]  1F 00 61 F1 FF FF```

According to the [dcf file](https://github.com/Linkeway/BIRL_modular_robot/blob/master/birl_module_canopen/config/birl_copley.dcf) I used to configure canopen_motor_node, RPDO2 is mapped to 0x6040(16bit) and 0x6060(8bit). So COB-id 301 should be transmitting 24 bits. Is it a problem from the dcf interpreter?

I'm using **indigo and Ubuntu14.04**.

Any help will be much appreciated.

gavanderhoorn, 2016-12-29:
Is this a cross-post from ROS Answers: [canopen_motor_node doesn't with dcf configuration](http://answers.ros.org/question/250917)?

Linkeway, 2016-12-29:
Sorry for that. I think it's better to post it here. I deleted it in answer.org.
I found the problem. I used **dcf_overlay** in yaml file. That's why I get the wrong PDO message.

closed 2016-12-29 by Linkeway

ipa-mdl, 2016-12-29:
> I found the problem.

👍 

`dcf_overlay` is meant for branching the configuration for different devices, especially motions settings like limits and ramp selection. PDO mapping should better be done in the DCF itself.