ros-industrial/ros_canopen #213: Sending Messages

Milliau, 2017-01-10:
Hi,
I'm new at ROS and CAN. 
I want to sent Signals with an USB CAN Adapter from Peak. My signal must send by an ID to an gateway. Which of the Interface I should use for it, is there an example code, for sending messages.

Thanks for your help

ipa-mdl, 2017-01-10:
You can use `socketcan_interface` classes in your code or you can use the ROS interfaces (topics) from `socketcan_bridge`.

Milliau, 2017-01-10:
thanks for your answer i tried to start the to_socketcan.test but it just start the master, i thin it's not sendin messages, also the LED on my USB-CAN Adapter is not blinking. Do you know if the to_socketcan-test works?

ipa-mdl, 2017-01-10:
`to_socketcan.test` is not using an actual CAN interface, it is just a unit test.

Milliau, 2017-01-10:
is there an exisiting code with an actual CAN interface with sends messages from a topic to CAN?

ipa-mdl, 2017-01-10:
Just spin up one of the nodes: http://wiki.ros.org/socketcan_bridge

Milliau, 2017-01-10:
thanks for your help

closed 2017-01-10 by Milliau