ros-industrial/ros_canopen #214: Failed to send message

Milliau, 2017-01-18:
I've written an publisher to use the topic to socketcan.cpp, when i start the launch file i get an error:

> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.20
>  * /topic_to_/can_device: can0
> 
> NODES
>   /
>     send (socketcan_bridge/SendMessage)
>     topic_to_ (socketcan_bridge/topic_to_socketcan_node)
> 
> auto-starting new master
> process[master]: started with pid [9024]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 7bcccfb0-dd8f-11e6-8c01-94659cd36e27
> process[rosout-1]: started with pid [9038]
> started core service [/rosout]
> process[topic_to_-2]: started with pid [9045]
> process[send-3]: started with pid [9053]
> [ INFO] [1484751901.777213075]: Successfully connected to can0.
> [ERROR] [1484751901.991464081]: Failed to send message: 80001337#0001020304050607.
> [ERROR] [1484751902.091409462]: Failed to send message: 80001337#0001020304050607.

```
#include "ros/ros.h"
#include <can_msgs/Frame.h>
#include <socketcan_interface/socketcan.h>
#include <socketcan_bridge/topic_to_socketcan.h>

int main(int argc, char **argv)
{
	ros::init(argc,argv, "talker");
	ros::NodeHandle nh;
	
	ros::Publisher publisher_ = nh.advertise<can_msgs::Frame>("sent_messages", 10);
	
	ros::Rate loop_rate(10);
	
	int count = 0;
	while(ros::ok())
	{
  can_msgs::Frame msg;
  msg.is_extended = true;
  msg.is_rtr = false;
  msg.is_error = false;
  msg.id = 0x1337;
  msg.dlc = 8;
  for (uint8_t i=0; i < msg.dlc; i++)
  {
    msg.data[i] = i;
  }

  msg.header.frame_id = "0";  // "0" for no frame.
  msg.header.stamp = ros::Time::now();

  // send the can_frame::Frame message to the sent_messages topic.
  publisher_.publish(msg);
  
  ros::spinOnce();
  loop_rate.sleep();
  count++;
 

}
return 0;
}
```
And this is the launch file i use
```
<launch>

  <node pkg="socketcan_bridge" type="topic_to_socketcan_node" name="topic_to_" output="screen">
          <param name="can_device" value="can0" />
        </node>
   <node pkg="socketcan_bridge" type="SendMessage" name="send"  />
 
</launch>
```

I can't find the problem, maybe someone has an idea whats the problem

ipa-mdl, 2017-01-18:
I have never been able to test it with extended IDs.
Do you get some out with `candump  can0`?

ipa-mdl, 2017-01-18:
Does your CAN NIC support extende IDs?

ipa-mdl, 2017-01-18:
Works with vcan:

```
rostopic pub -r 10 /sent_messages can_msgs/Frame 'header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
id: 0x1337
is_rtr: false
is_extended: true
is_error: false
dlc: 8
data: "\x00\x01\x02\x03\x04\x05\x06\x07"
'
```
results in
```
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
  vcan0  00001337   [8]  00 01 02 03 04 05 06 07
```

iwanders, 2017-01-18:
I searched in vain for a way to create a `vcan` device that only accepts standard frames to test this. I'm under the impression that [vcan](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/drivers/net/can/vcan.c?id=refs/tags/v4.9.4#n84) always accepts all input.

Milliau, 2017-01-18:
Thanks for your ideas, i will try it tommorow. I don't know The Different between can and vcan.

Milliau, 2017-01-19:
if I try this ` cansend can0 00001337#0001020304050607` I get write: Network is down.
if I try` rosrun socketcan_bridge topic_to_socketcan_node _can_device:=vcan0` it is successfully connected to the can0. But when I sart sending a message like ipa_mdl write it I get also the error Failed Message.

Can someone explain me the differtent between using vcan and can?
Does your CAN NIC support extended IDs? I think that this is the problem but i don't know how i can send an not extended ID. I changed msg.is_extended to flase. But i get the error: Invalid frame from topic. I read at Wikipedia about the base and extended format. 

Can somebody gives my an example of sending an base can message? Thanks

iwanders, 2017-01-19:
> Can someone explain me the differtent between using vcan and can?

In Linux the standard system to deal with CAN messages is [SocketCAN](https://en.wikipedia.org/wiki/SocketCAN). SocketCAN provides a standard mechanism to access CAN devices. Your hardware driver creates an interface between Linux' SocketCAN system and the device. If you have a hardware device this is often available as `can0`. For testing purposes, a virtual device can be created with the `vcan` kernel module. Think of this as a driver that creates a virtual device, this requires no hardware but still allows you to interact with it as if it were a real hardware device. This is often used for testing.

> Does your CAN NIC support extended IDs? I think that this is the problem but i don't know how i can send an not extended ID. I changed msg.is_extended to false. But i get the error: Invalid frame from topic.

Standard CAN frames ([explanation here](https://en.wikipedia.org/wiki/CAN_bus#Base_frame_format)) use 11 bits for the CAN ID. With 11 bits there are 2048 possible can ID's, if you want send a can message that is valid you have to ensure the CAN ID is below 2048. In your example you used `0x1337` as the CAN ID, which equals 4919 in decimal, that is above 2048 and would be an invalid frame if `is_extended` is false. Try an ID like `0xFF` (=255) and setting `is_extended` to false. Example with cansend: `cansend can0 000000FF#0001020304050607`.

> Network is down.

It could be that you have to manually bring up the device. This enables the device and configures it for a specific bitrate. See [this](http://elinux.org/Bringing_CAN_interface_up#Native_Interfaces) for an example how to do that. The extensive documentation on this can be found in the kernel [documentation](https://www.kernel.org/doc/Documentation/networking/can.txt).

ipa-mdl, 2017-01-19:
Which CAN device/dongle do you use?

Milliau, 2017-01-19:
I use an USB-CAN Adapter from PEAK.

closed 2017-01-25 by Milliau