ros-industrial/ros_canopen #223: issues while trying the CAN exaamples

Ayushas2427, 2017-02-17:
I am new to the topic of ROS CAN interfaces and trying to implement the following two package sources
https://github.com/ros-industrial/ros_canopen
and 
https://github.com/JCube001/socketcan-demo

Unfortunately, i m not able to implement any of them.
In first link of "ros_canopen", i am getting the following error while performing "catkin_make"

[image]
 
in the second link, Though everything was ok till i did "catkin_make' but the rosrun is showing the error of "package not found". I did the "source devel/setup.bash" which also did not give any error. So, why the rosrun is not able to be implemented? 

Thank you in advance.
Regards

enricovillagrossi, 2017-02-17:
To solve the first problem you can try to install libmuparser with:
sudo apt-get install  libmuparser-dev

ipa-mdl, 2017-02-17:
Even better: use rosdep install..
(Or try the released version)

Ayushas2427, 2017-02-17:
Thank you @enricovillagrossi. Your solution rectified my error.
@ipa-mdl , Thank you for your reply as well , since rosdep gets the required dependencies, will the help me to rectify my second issue as well?

Ayushas2427, 2017-02-17:
Probably, here is where i am getting the error. When i perform "roscd <package name> " , everything goes fine, but when i did the same, for "socketcan_demo" package (which i explained in my second issue) , then it shows the error as "roscd: No such package/stack 'socketcan_demo'".

It means that my system is not able to find the package. What can i infer from this? and how can i resolve such issues?

Regards

ipa-mdl, 2017-02-17:
`socketcan-demo` (no catkin/ROS packages)  and 'ros_canopen`  (catkin packages) are unrelated.

Ayushas2427, 2017-02-17:
ok. Probably I was mixing two separate concepts. Thank you for rectifying

ipa-mdl, 2017-03-20:
seems to be off topic -> close

closed 2017-03-20 by ipa-mdl