ros-industrial/ros_canopen #231: CAT vehicle teleop mode error

Ayushas2427, 2017-04-09:
I have used the tutorials of the [CAT Vehicle](http://cps-vo.org/node/31792) . i simulated the vehicle on gazebo and the vehicle performed well.

Now, i used the teleop mode to move the vehicle.
The TurtleBot teleop node is publishing Twist messages to /cmd_vel_mux/input/teleop ,and to convert them into /catvehicle/cmd_vel.

The following command is used for conversion:
_$ rosrun topic_tools relay /cmd_vel_mux/input/teleop /catvehicle/cmd_vel_

But the vehicle is not responding. May i please know where is the error?
Though,the vehicle is responding well with the "hard left turn" as shown in the [tutorial](http://cps-vo.org/node/26603).

Kindly help.
Best Regards

ipa-mdl, 2017-04-09:
How does this relate to `ros_canopen`?

Ayushas2427, 2017-04-09:
Oh, I am very sorry. By mistake i posted it at the wrong location. Kindly take it as the first and final error from my side.

Shall remove it from here and will post it at a suitable location.

Best Regards

Ayushas2427, 2017-04-09:
Closing the question. Re-posted it to the following link.
https://github.com/ros-industrial/industrial_experimental/issues/12

Regards

closed 2017-04-09 by Ayushas2427