ros-industrial/ros_canopen #236: Error at robot_layer.cpp

epsilon425, 2017-05-12:
Hi I'm trying to build the canopen package but I keep on getting errors concerning the
robot_layer.cpp. 

[ 97%] Building CXX object ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor_node.dir/src/robot_layer.cpp.o
/home/paul/catkin_ws/src/ros_canopen/canopen_motor_node/src/robot_layer.cpp: In member function ‘virtual bool RobotLayer::prepareSwitch(const std::__cxx11::list<hardware_interface::ControllerInfo>&, const std::__cxx11::list<hardware_interface::ControllerInfo>&)’:
/home/paul/catkin_ws/src/ros_canopen/canopen_motor_node/src/robot_layer.cpp:378:63: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
             if(g_interface_mapping.hasConflict(controller_it->hardware_interface

as you can see, hardware_interface cannot be referenced. Sorry for the trouble and thank you for your time!

ipa-mdl, 2017-05-12:
The issue tracker is no support forum, please use http://answers.ros.org for this purpose.

It looks like you are trying to build the `jade-devel` branch of `ros_canopen` against the `indigo` version of `ros_control`. The latter introduced an API break from `indigo` to `jade`, so please use corresponding versions.

epsilon425, 2017-05-15:
thank you and I apologize for misusing the issue tracker

closed 2017-05-17 by epsilon425