ros-industrial/ros_canopen #237: PCAN-miniPCIe send errors

reinzor, 2017-05-12:
We experience errors sometimes with our can drivers based on the socketcan_interface i.c.m. PCAN-miniPCIe modules. Only receiving seems to work fine but if we are also sending using `can_driver_.send(f)`, sending fails sometimes.

We have tested with the kernel driver: PEAK_PCI and the vendor driver from [PCAN](http://www.peak-system.com/fileadmin/media/linux/index.htm) when compiled with NETDEV. 

Does somebody have experience with these kind of issues? Thanks!

ipa-mdl, 2017-05-12:
Could be related to https://github.com/ros-industrial/ros_canopen/issues/102
If you send messages too fast, the socket queue cold overflow.

Do you get an error code? 
Is `ip -details -statistics link show can0` showing errors?

We had troubles with the kernel driver and the PEAK-cPCI cards, but the vendor driver in `NETDEV` mode worked for us.

reinzor, 2017-05-12:
`
11: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 100 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  sja1000: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         
    RX: bytes  packets  errors  dropped overrun mcast   
    0          0        0       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    137820     9188     0       0       0       0   
`

I can simply reproduce the problem now indeed by sending too much messages over the CANbus. However, when I stop sending, the socketcan_interface does not recover unless I restart the application. 

I am testing using the socketcan_bridge:
`rosrun socketcan_bridge socketcan_bridge_node _can_device:=can0`

`rostopic pub /sent_messages can_msgs/Frame {} -r RATE`

When I set the rate to 10khz for example, the program says: 
`[ERROR] [1494600652.166214227]: Failed to send message: 0#.`

However, when I reduce the rate afterwards to 10 hz for example, it does not recover. Is this expected behaviour?

ipa-mdl, 2017-05-12:
The details look fine, everthing is okay.

> it does not recover. Is this expected behaviour?

Unfortunately, yes.
`can_driver_.recover()` needs to be called explicitly.

As a workaround, you can increase the queue length (`sudo ip link set can1 txqueuelen 15`) to work-around the error.

reinzor, 2017-05-14:
Thanks for the response. I will try with an increased txqueue length and if it fails, try to automatically recover using the recover method. I will test this tomorrow and send an update on the issue whether this has worked for me.

reinzor, 2017-05-15:
Increasing the `txqueuelen` did help. Also added a `can_driver_.recover()` when the sending fails. Now the driver recovers again when we disconnect and connect the CAN cable or if the received does not acknowledge for some time period. Thanks for all your help; I will close the issue.

closed 2017-05-15 by reinzor

GeoDTN, 2018-02-27:
Dear All,
I am because I have similar issue with Rx. I checked with canalyser, there is no frame error at physical layer.
And there is no error with TX. When I issue ifconfig can1,  I get error on received frames only.
I was testing transmission between nodes. When this error happens, the sender retransmits from the beginning of the file.
How can I solve, it is urget project and I am worried. I tried your txqueue increment, but I think it is for Tx error and it didnt help me. Can you suggest similar approach for Rx error on embedded arm iMX6 Modules?
Thank you in advance.

GeoDTN, 2018-02-28:
[image]
[image]

ipa-mdl, 2018-03-01:
@GeoDTN: Please do not hijack unrelated, closed issues. If you think this is an error in socketcan_interface, just open an issue here. Otherwise, I would suggest to ask for help at ROS answers.
The screenshots you have posted do not help that much in identifying the problem.

GeoDTN, 2018-03-01:
Thank you.Sorry if I made a mistake.
The point is we have frame errors sometimes(especially when we transmit
relatively longer files). We analysed the frames at hardware level at
receiver with Canalyzer and Can case. But at hardware level there is no
error when the error happens.Then when we issued ifconfig can1, we see
frame error on received frames.There is no error on transmitted frames. I
guessed(not sure) the source of problem is kernel of the embedded module as
we proved there is no error on received frames at hardware level and
transmitted frames.
Thank you in advance for help.
Kind regards,