ros-industrial/ros_canopen #240: Errors with socketcan_bridge debian package

Timple, 2017-07-10:
We get an error running the debian package.

```bash
$ rosrun socketcan_bridge topic_to_socketcan_node 
[ INFO] [1499686326.788331478]: Successfully connected to can0.
/opt/ros/kinetic/lib/socketcan_bridge/topic_to_socketcan_node: symbol lookup error: /opt/ros/kinetic/lib/socketcan_bridge/topic_to_socketcan_node: undefined symbol: _ZN16socketcan_bridge16TopicToSocketCANC1EPN3ros10NodeHandleES3_N5boost10shared_ptrIN3can15DriverInterfaceEEE
```

We use the native CAN on the A20-OLinuXino-LIME2-eMMC board. (ARM based).

Version of the debian:
```bash
$ dpkg --status ros-kinetic-socketcan-bridge 
Package: ros-kinetic-socketcan-bridge
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 575
Maintainer: Mathias Lüdtke <mathias.luedtke@ipa.fraunhofer.de>
Architecture: armhf
Version: 0.7.5-0xenial-20170614-111241-0800
Depends: libboost-system1.58.0, libboost-thread1.58.0, libc6 (>= 2.9), libgcc1 (>= 1:3.5), libstdc++6 (>= 5.2), ros-kinetic-can-msgs, ros-kinetic-roscpp, ros-kinetic-socketcan-interface
Description: Provides nodes to convert messages from SocketCAN to a ROS Topic and vice versa.
Homepage: http://wiki.ros.org/socketcan_bridge
```

Unfortunately (and maybe related) socketcan_bridge also does not compile:
```bash
Errors     << socketcan_bridge:symlink /home/noblean/ros/kinetic/system/logs/socketcan_bridge/build.symlink.003.log                                                                                                                          
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/catkin_tools/execution/stages.py", line 174, in function_proxy
    return function(logger, event_queue, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/jobs/catkin.py", line 291, in link_devel_products
    source_hash = md5(open(os.path.realpath(source_file)).read().encode('utf-8')).hexdigest()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc8 in position 25: ordinal not in range(128)

Stage `symlink` failed with arguments:
  package: {'package_format': 2, 'exports': [], 'buildtool_export_depends': [], 'exec_depends': [<catkin_pkg.package.Dependency object at 0xb61e3930>, <catkin_pkg.package.Dependency object at 0xb61e3f90>, <catkin_pkg.package.Dependency object at 0xb61e3cc0>], 'name': 'socketcan_bridge', 'filename': '/home/noblean/ros/kinetic/system/src/socketcan_bridge/package.xml', 'replaces': [], 'version_abi': None, 'build_export_depends': [<catkin_pkg.package.Dependency object at 0xb61e3960>, <catkin_pkg.package.Dependency object at 0xb61e39c0>, <catkin_pkg.package.Dependency object at 0xb61e3c60>], 'doc_depends': [], 'test_depends': [<catkin_pkg.package.Dependency object at 0xb61e3c30>, <catkin_pkg.package.Dependency object at 0xb61e3ed0>, <catkin_pkg.package.Dependency object at 0xb61e3900>], 'maintainers': [<catkin_pkg.package.Person object at 0xb62c4350>], 'version': '0.7.5', 'build_depends': [<catkin_pkg.package.Dependency object at 0xb61d81e0>, <catkin_pkg.package.Dependency object at 0xb61d8210>, <catkin_pkg.package.Dependency object at 0xb61d8240>], 'urls': [<catkin_pkg.package.Url object at 0xb62c4410>, <catkin_pkg.package.Url object at 0xb62c4390>, <catkin_pkg.package.Url object at 0xb62c4530>], 'authors': [<catkin_pkg.package.Person object at 0xb62c44b0>], 'licenses': ['BSD'], 'buildtool_depends': [<catkin_pkg.package.Dependency object at 0xb61d8090>], 'conflicts': [], 'description': u'Provides nodes to convert messages from SocketCAN to a ROS Topic and vice versa.'}
  metadata_path: /home/noblean/ros/kinetic/system/.catkin_tools/profiles/default
  devel_manifest_path: /home/noblean/ros/kinetic/system/.catkin_tools/profiles/default/packages/socketcan_bridge
  prebuild: False
  package_path: socketcan_bridge
  dest_devel_path: /home/noblean/ros/kinetic/system/devel
  source_devel_path: /home/noblean/ros/kinetic/system/devel/.private/socketcan_bridge
.............................................................................................................................................................................................................................................
Failed     << socketcan_bridge:symlink           [ Exited with code 3 ]                                                                                                                                                                      
Failed    <<< socketcan_bridge                   [ 2 minutes and 34.2 seconds ]                    
```
Compiling socketcan_interface from source works fine.

ipa-mdl, 2017-07-10:
Thanks for reporting!

Regarding the rosrun issue:
* What's the output of
`ldd -r /opt/ros/kinetic/lib/socketcan_bridge/topic_to_socketcan_node`?
* Which `socketcan_interface` version are you using?
* I will try it on a Raspberry PI if I find some time, in a Docker container everything looks correct.

Your build error looks like https://github.com/catkin/catkin_tools/pull/399, a release is still pending.
It should work with `catkin_make`.

Timple, 2017-07-10:
- Output:

```bash
$ ldd -r /opt/ros/kinetic/lib/socketcan_bridge/topic_to_socketcan_node
	linux-vdso.so.1 =>  (0xbeb59000)
	libtopic_to_socketcan.so => /home/noblean/ros/kinetic/system/devel/lib/libtopic_to_socketcan.so (0xb6eca000)
	libroscpp.so => /opt/ros/kinetic/lib/libroscpp.so (0xb6da6000)
	librosconsole.so => /opt/ros/kinetic/lib/librosconsole.so (0xb6d72000)
	libboost_system.so.1.58.0 => /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.58.0 (0xb6d2c000)
	libboost_thread.so.1.58.0 => /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.58.0 (0xb6d01000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6cdc000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6bce000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6b56000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6b2e000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6a42000)
	/lib/ld-linux-armhf.so.3 (0x7f58e000)
	libroscpp_serialization.so => /opt/ros/kinetic/lib/libroscpp_serialization.so (0xb6a2e000)
	libsocketcan_interface_string.so => /opt/ros/kinetic/lib/libsocketcan_interface_string.so (0xb6a1a000)
	librostime.so => /opt/ros/kinetic/lib/librostime.so (0xb69ed000)
	libxmlrpcpp.so => /opt/ros/kinetic/lib/libxmlrpcpp.so (0xb69c9000)
	libcpp_common.so => /opt/ros/kinetic/lib/libcpp_common.so (0xb69b3000)
	libboost_filesystem.so.1.58.0 => /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.58.0 (0xb6991000)
	librosconsole_log4cxx.so => /opt/ros/kinetic/lib/librosconsole_log4cxx.so (0xb696f000)
	librosconsole_backend_interface.so => /opt/ros/kinetic/lib/librosconsole_backend_interface.so (0xb695d000)
	liblog4cxx.so.10 => /usr/lib/arm-linux-gnueabihf/liblog4cxx.so.10 (0xb6817000)
	libboost_regex.so.1.58.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.58.0 (0xb6760000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6749000)
	libconsole_bridge.so.0.2 => /usr/lib/arm-linux-gnueabihf/libconsole_bridge.so.0.2 (0xb6735000)
	libapr-1.so.0 => /usr/lib/arm-linux-gnueabihf/libapr-1.so.0 (0xb6708000)
	libaprutil-1.so.0 => /usr/lib/arm-linux-gnueabihf/libaprutil-1.so.0 (0xb66df000)
	libicui18n.so.55 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.55 (0xb6545000)
	libicuuc.so.55 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.55 (0xb6428000)
	libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0xb6414000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6401000)
	libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0xb63c2000)
	libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0xb6399000)
	libicudata.so.55 => /usr/lib/arm-linux-gnueabihf/libicudata.so.55 (0xb4ad2000)
undefined symbol: _ZN16socketcan_bridge16TopicToSocketCANC1EPN3ros10NodeHandleES3_N5boost10shared_ptrIN3can15DriverInterfaceEEE	(/opt/ros/kinetic/lib/socketcan_bridge/topic_to_socketcan_node)
```

- ```bash
$ dpkg --status ros-kinetic-socketcan-interface 
Package: ros-kinetic-socketcan-interface
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 425
Maintainer: Mathias Lüdtke <mathias.luedtke@ipa.fraunhofer.de>
Architecture: armhf
Version: 0.7.5-0xenial-20170529-122847-0800
Depends: libboost-system1.58.0, libboost-thread1.58.0, libc6 (>= 2.9), libconsole-bridge0.2v5, libgcc1 (>= 1:3.5), libstdc++6 (>= 5.2), libboost-all-dev, linux-libc-dev, ros-kinetic-class-loader
Description: This package contains a generic CAN interface description with helpers for filtering and driver implementation.
 Further a socketcan implementation based on boost::asio is included.
Homepage: http://wiki.ros.org/socketcan_interface
```

- And true, we use catkin tools.

ipa-mdl, 2017-07-10:
Your output contains:
> libtopic_to_socketcan.so => /home/noblean/ros/kinetic/system/devel/lib/libtopic_to_socketcan.so

It is overlayed by your workspace. The `sockecan_interface` code is templated, so it might not match if you are using other compiler settings.

Please try to run it in a clean shell with only `/opt/ros/kinetic/setup.bash` sourced.
Or rebuild the workspace with `catkin_make`.

ipa-mdl, 2017-08-04:
@Timple: Did you solve it?

Timple, 2017-12-01:
Yes, this is solved now. Indeed was an overlay issue. I did not expect the debian versions to use the manually compiled libs.

closed 2017-12-01 by Timple