ros-industrial/ros_canopen #251: Support Kvaser linuxcan

JWhitleyAStuff, 2017-09-23:
This is more of a discussion topic than a true issue but it's also somewhat of a feature request (that, if the community is OK with, I might implement myself). Wanted to get feedback on implementing can::KvaserInterface in addition to the existing can::SocketCANInterface. Even though Kvaser Leaf Lights are supported natively under the `kvaser_usb` kernel module, there are many other Kvaser devices which are only supported with Kvaser's linuxcan API. Thoughts?

ipa-mdl, 2017-09-23:
In the early versions `ros_canopen` included an [additional driver implementation for PEAK devices](https://github.com/ros-industrial/ros_canopen/tree/62807c28c540697f4bd6cf57202f8871bd4ce784/ipa_peak_driver).
It has been removed because the socketcan interface provides concurrent access and does not need external libraries. Eventually, it became the default option.

However, the code is prepared for addtional drivers that implement can::DriverInterface.
In addition a plugin mechanism was added (currenty not supported by `socketcan_bridge` #252).

It shouldn't be too hard to implement a Kvaser driver. I am available for advice and review :)
The driver should definately live in its own package, I am undecided if it should be included in this repository. For now I would suggest to keep it separated.

JWhitleyAStuff, 2017-09-25:
Thanks for the feedback, @ipa-mdl. I'll fork this repo and first try to add the plugin capability. I'll submit a PR for that and then add the packages (kvaser_interface and kvaser_bridge). Once I'm finished, I'll submit a PR for the new plugin and you and any other maintainers can decide then if you want it in this package.

ipa-mdl, 2017-09-26:
Great!
I will try to solve #252, so no need for `kvaser_bridge` then.

JWhitleyAStuff, 2017-11-21:
Because we already had a partial solution , we didn't want this package to rely on the linuxcan SDK, and for several other reasons, we made our own package for this purpose. See https://github.com/astuff/kvaser_interface

Thanks for your help!

closed 2017-11-21 by ipa-mdl

gavanderhoorn, 2017-11-21:
@JWhitleyAStuff wrote:
> we didn't want this package to rely on the linuxcan SDK

wouldn't just your pkg (which would provide the plugin) depend on linuxcan in the proposed scheme?