ros-industrial/ros_canopen #255: Topic to SocketCan message does not get sent to bus if using rostopic echo

h5tahir, 2017-10-27:
I wrote a simple node that publishes a frame on the "sent_messages" topic to a radar connected on the CANBUS. 

What I think happens?: It seems the message gets put on the bus if there is no other subscribers on the bus? 
Why?: 'rostopic echo /sent_messages' displays the frame in console, however the frame is not actually picked up the msgCallback in topic_to_socketcan.cpp. If you remove the subscription from 'rostopic echo ...' the msgCallback picks up the message.

Is this something I am doing wrong, or is this supposed to happen?

ipa-mdl, 2017-10-27:
Can you upload your code to https://gist.github.com/?

h5tahir, 2017-10-27:
This is the link to the node I created:
https://gist.github.com/h5tahir/439b5aafed3811113017d6e2ac5a7de9

You can duplicate this by the following steps:
1. Put a comment in msgCallback function in topic_to_socketcan.cpp and build
2. Run roscore
3. Run rosrun socketcan_bridge socketcan_bridge_node (another terminal)
4. Run rostopic echo /sent_messages (another terminal)
5. Run my node (code above - another terminal) - can see the comment is not printed
6. Stop rostopic echo process
7. Run my node - can see the comment is printed

h5tahir, 2017-10-27:
It seems:
`    while (configPublisher.getNumSubscribers() < 1)
    {
        // waiting for there to be a subscriber and the publisher to finish setting up
}`
is not the best way of waiting for the publisher to be fully set up as the message is published before all subscribers are properly attached, i guess. If I change this to getNumSubscribers() < 2, things work. So, I believe it is just a problem with the way I wrote my node. 

Thanks. Going to close this issue.

closed 2017-10-27 by h5tahir

ipa-mdl, 2017-10-27:
In addtion your code exits immediately, but publish() does not block until the message was passed to all subscribers.