ros-industrial/ros_canopen #256: canopen_motor_node segfaults on eds file.

Timple, 2017-11-17:
Upon parsing the eds file the canopen_motor_node segfaults.

I traced the issue back to the `parse_objects` function in the `objdict.cpp` file of canopen_master.

The issue seems to be with the 2007subX values in the eds file.

Parsing the eds file would be preffered, otherwise giving a usefull hint instead of a segfault would be a nice second.

EDS file attached.
[Engel_Drive_1_9.eds.txt](https://github.com/ros-industrial/ros_canopen/files/1482000/Engel_Drive_1_9.eds.txt)

ipa-mdl, 2017-11-17:
It is caused by `2007sub1`  which has datatype 1 (1bit booelan), but this is not supported by `ros_canopen`.
I would suggest to change the datatype to 5 (UNSIGNED8) in the EDS.

However, the parser needs some better error handling :-/
Not sure if we should just skip unsupported entries or crash with a proper message.

ipa-mdl, 2017-11-17:
With #258 it will still fail to load, but it will read:
> terminate called after throwing an instance of 'canopen::ParseException'
  what():  Cannot process 0x2007: Cannot process 0x2007sub1: std::bad_cast

Timple, 2017-11-17:
Nice, we'll try putting the datatype to 5

Timple, 2017-11-20:
> terminate called after throwing an instance of 'canopen::ParseException'
> what(): Cannot process 0x2007: Cannot process 0x2007sub1: std::bad_cast

This helps a lot by pinpointing the issue to the 0x2007sub1 item, this took me a while. But then it does not hint a solution.

Would something like this be possible:
> terminate called after throwing an instance of 'canopen::ParseException'
> what(): Cannot process 0x2007: Cannot process 0x2007sub1: std::bad_cast, type is 0x01 which is not supported, supported types are: 0x05 0x09

ipa-mdl, 2018-01-08:
I have now altered the output to:

> terminate called after throwing an instance of 'canopen::ParseException'
what(): Cannot process 0x2007: Type of  0x2007sub1 does not match or is not supported

Timple, 2018-03-13:
Works for me!

closed 2018-03-13 by Timple