ros-industrial/ros_canopen #257: Could not find resource 'left_wheel' in 'hardware_interface::VelocityJointInterface'.

Timple, 2017-11-17:
We get an error trying to operate our CANOpen motors. The velocity interface does not seem to be registered.

```bash
[ERROR]: Exception thrown while initializing controller base_controller.
Could not find resource 'left_wheel' in 'hardware_interface::VelocityJointInterface'.
[ERROR]: Initializing controller 'base_controller' failed
```

In simulation (gazebo) and on other hardware (etherCAT) we have the same robot running including the same urdf and the same ros_control.yaml file. So the parameters should be correct.

I have the feeling that canopen_motor_node does not 'know' that our motor supports velocity setpoints. Is there something we're doing wrong?

Relevant pieces of config:

**ros_control.yaml**
```yaml
joint_state_controller:
  type: joint_state_controller/JointStateController
  publish_rate: 50

base_controller:
  type        : "diff_drive_controller/DiffDriveController"
  open_loop   : true
  left_wheel  : 'left_wheel'
  right_wheel : 'right_wheel'

left_wheel_controller:
  type: velocity_controllers/JointVelocityController
  joint: left_wheel
  required_drive_mode: 3
```

**canopen.yaml**
```yaml
canopen_motor_chain_node:
  bus:
    device: can0
  sync:
    interval_ms: 100
    overflow: 0
  nodes:
    node2:
      id: 2
      name: left_wheel
      eds_pkg: "ourbot_control"
      eds_file: "config/HBI3760-4/Engel_Drive_1_9.eds"
    node3:
      id: 3
      name: right_wheel
      eds_pkg: "ourbot_control"
      eds_file: "config/HBI3760-4/Engel_Drive_1_9.eds"

hardware_interface:
  joints:
    - left_wheel
    - right_wheel

joint_group_velocity_controller:
  joints:
    - left_wheel
    - right_wheel
  required_drive_mode: 3
  type: velocity_controllers/JointGroupVelocityController
```

**ourbot.urdf**
```xml
 <transmission name="left_wheel_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="left_wheel">
        <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </joint>
    <actuator name="left_wheel_actuator">
        <mechanicalReduction>7</mechanicalReduction>
        <hardwareInterface>VelocityJointInterface</hardwareInterface>
    </actuator>
</transmission>

<transmission name="right_wheel_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="right_wheel">
        <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
    </joint>
    <actuator name="right_wheel_actuator">
        <mechanicalReduction>7</mechanicalReduction>
        <hardwareInterface>VelocityJointInterface</hardwareInterface>
    </actuator>
</transmission>
</robot>
```

ipa-mdl, 2017-11-17:
Your EDS lacks the mandatory object 0x6052 (supported drive modes), so `canopen_402` does not offer these interfaces.
You can add it with a default value, the (non-cmpliant) device does not actually need to support reading it.

Timple, 2017-11-23:
Do you perhaps mean 0x6502? I found more references for this value than 0x6052.

Could you give a more elaborate example? Simply stating: 
```

[6502]
ParameterName=Supported Drive Modes
ObjectType=0x07
DataType=0x0007
AccessType=ro
PDOMapping=0
```
And adding it to the `[OptionalObjects]` list doesn't work. 

The canopen motor node then gives:
```
[ INFO] [1511448070.037196898]: Using fixed control period: 0.100000000
[ INFO] [1511448072.377924813]: Initializing XXX
[ INFO] [1511448072.378936252]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1511448072.379573832]: Current state: 2 device error: system:0 internal_error: 0 (OK)
abort6502#0, reason: Object does not exist in the object dictionary.
Could not process message
illegal tranistion 0 -> 2
[ INFO] [1511448074.425802690]: Current state: 2 device error: system:125 internal_error: 0 (OK)
[ INFO] [1511448074.426379839]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[ INFO] [1511448074.426534218]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[ INFO] [1511448074.426763768]: Current state: 0 device error: system:0 internal_error: 0 (OK)
```

ipa-mdl, 2017-11-23:
> Do you perhaps mean 0x6502? I found more references for this value than 0x6052.

Yes, sry.. These look all the same ;)

Timple, 2017-11-24:
No worries, any idea why adding this is not enough?

Simply stating: 
```
[6502]
ParameterName=Supported Drive Modes
ObjectType=0x07
DataType=0x0007
AccessType=ro
PDOMapping=0
```
And adding it to the `[OptionalObjects]` list.

It should be read from the eds and than registered as supported mode. But where this is happening is still a mystery for me.

Timple, 2017-11-24:
I've found [here](https://github.com/gw-cs-sd/sd-2017-BRIEF-Crandall/blob/dca77a3a6c64ef189cb442cb1384f99fc63723fa/src/schunk_robots/schunk_lwa4d/config/Schunk_0_63.dcf) that I could add a default value. (Sorry, I overlooked your statement about adding a default value I see now.) According to the definition [here](https://us.nanotec.com/products/manual/PD2C_CAN_EN/object_dictionary%252Fod_motion_0x6502.html/) a value of 15 would result in supporting _Profile Position mode_, _Velocity mode_, _Profile Velocity mode_ and _Torque mode_.

The resulting part in the eds file:
```
[6502]
ParameterName=Supported Drive Modes
ObjectType=0x07
DataType=0x0007
AccessType=ro
DefaultValue=0x15
PDOMapping=0
```

closed 2017-11-24 by Timple