ros-industrial/ros_canopen #259: joint_state_publisher publishes at 2hz while set at 10hz

Timple, 2017-12-01:
`joint_state_publisher` and the `diff_drive_controller` both run at 2hz while set to 10hz.

The `joint_state_publisher` starts at about 3-4hz but reduces to 2hz when the `diff_drive_controller` is also enabled.

This is also the frequency at which the canmotors are operated (I can hear the velocity changes at around 2hz)

Relevant configs:
```yaml
canopen_motor_chain_node:
  bus:
    device: can0

  sync:
    interval_ms: 100
    overflow: 0

  defaults:
    eds_file: "config/HBI3760-4/Engel_Drive_1_9_ROS_compatible.eds"
    dcf_overlay:
      "1800sub2": "1" # Only send PDO on sync message (Reduces chatter on CANbus)
      "1801sub2": "1"
      "1017": "1000" # heartbeat producer [ms]

  nodes:
    node2:
      id: 2
      name: left_wheel
      dcf_overlay:
        "607E": "1" # Reverse Polarity

    node3:
      id: 3
      name: right_wheel

  diagnostic_period: 1.0

hardware_interface:
  joints:
    - left_wheel
    - right_wheel
```

```yaml
joint_state_controller:
  type: joint_state_controller/JointStateController
  publish_rate: 10

base_controller:
  type        : "diff_drive_controller/DiffDriveController"
  open_loop   : false
  left_wheel  : 'left_wheel'
  right_wheel : 'right_wheel'
  publish_rate: 10.0               # default: 50
  publish_cmd : true

  wheel_radius        : 0.1
  wheel_separation    : 0.5
  base_frame_id       : 'base_link'
  enable_odom_tf      : false
  required_drive_mode : 3

left_wheel_velocity_controller:
  type: velocity_controllers/JointVelocityController
  joint: left_wheel
  required_drive_mode: 3

right_wheel_velocity_controller:
  type: velocity_controllers/JointVelocityController
  joint: right_wheel
  required_drive_mode: 3

```

* Changing bitrate of the can had no influence
* Changing the sync interval helped a little bit, could reach 4.7hz
* Setting PDO to asynchonous with a inhibit time of 20hz did not help
* The computer does not seem busy (htop <5%) (canbusload <2%)

ipa-mdl, 2017-12-01:
Your `interval_ms` is 100, so `canopen_motor_node` will run at 10Hz.
The publisher in `joint_state_publisher` is decoupled from this and schould run at `publish_rate: 10`Hz, I guess it's the same for the `DiffDriveController`.

By default `canopen_motor_node` will update the controllers at the sync rate.
E.g. the values cannot change faster..

How do you measure the rate?

Do you see proper SYNC timing in the candum?
Are there any SDO involved?

Have you mapped all [cyclic data](http://wiki.ros.org/canopen_402#Overview)?

Timple, 2017-12-02:
> Your interval_ms is 100, so canopen_motor_node will run at 10Hz.

Reducing the interval to 10ms did not make much of a difference. I do realise the data received would be more recent.

> How do you measure the rate?

Using rostopic hz. But besides this we can also see and hear for example the velocity go down in steps of 2hz due to the acceleration limit. So it does not seem to be a rostopic issue.

> Do you see proper SYNC timing in the candum?

Yes, if I remember correctly [update on monday]

> Are there any SDO involved?

I get the concept of using PDOs, but haven't be able to modify the data used by the PDOs using CANeds (version 3.6).
The candump looks like only PDOs are involved (looking at the canid) [update on monday]

> Have you mapped all cyclic data?

Not explicitly, but I think it is set correctly [update on monday]

ipa-mdl, 2017-12-02:
> The publisher in joint_state_publisher is decoupled from this and schould run at publish_rate: 10Hz, I guess it's the same for the DiffDriveController.

Actually, this is wrong. The publish is triggered by the update, so it cannot get published faster than the update loop runs.

So it looks like something is wrong with your update loop.
Let's see what you can find in the candump.

Timple, 2017-12-04:
With a single node attached this is the output of canopen_motor_node

```bash
[ INFO] [1512373690.698719338]: Using fixed control period: 0.100000000
[ INFO] [1512373695.666165744]: Initializing XXX
[ INFO] [1512373695.667354691]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1512373695.668143565]: Current state: 2 device error: system:0 internal_error: 0 (OK)
EMCY: 83#B081010000000000
EMCY: 83#A081010000000000
EMCY: 83#0000000000000000
EMCY: 83#0000000000000000
```

And this is the outputof the candump after calling `/driver/init`
And rosrun controller_manager controller_manager spawn joint_state_controller
```bash
$ candump can0 -td -dex
 (000.000000) TX 000 [2]  82 03                   -> reset communication node 3
 (000.014403) RX 703 [1]  00                      -> node 3 switched to pre-operational
 (000.016868) RX 083 [8]  B0 81 01 00 00 00 00 00 -> Emergency Telegram: CAN stopped
 (000.019242) RX 083 [8]  A0 81 01 00 00 00 00 00 -> Emergency Telegram: CAN reset communication
 (000.116431) TX 603 [8]  2B 17 10 00 E8 03 00 00 -> Write request: Producer Heartbeat Time: 1000ms
 (000.122822) RX 583 [8]  60 17 10 00 00 00 00 00 -> Write response
 (000.225093) TX 603 [8]  40 41 60 00 00 00 00 00 -> Read request:  Statusword
 (000.231289) RX 583 [8]  4B 41 60 00 18 02 8F AF -> Read responce: Statusword: Fault
 (000.332584) TX 603 [8]  40 61 60 00 00 00 00 00 -> Read request:  Modes of Operation Display
 (000.339024) RX 583 [8]  4F 61 60 00 01 BC 00 8F -> Read responce: Modes of Operation Display: Profile Position Mode
 (000.441702) TX 603 [8]  40 64 60 00 00 00 00 00 -> Read request:  Position Actual Value
 (000.448119) RX 583 [8]  43 64 60 00 FF 11 00 00 -> Read responce:
 (000.550335) TX 603 [8]  40 40 60 00 00 00 00 00 -> Read request:  Controlword
 (000.556649) RX 583 [8]  4B 40 60 00 00 00 8F AF -> Read responce: Controlword: Fault
 (000.658200) TX 603 [8]  40 60 60 00 00 00 00 00 -> Read request:  Modes of Operation
 (000.664499) RX 583 [8]  4F 60 60 00 00 00 8F AF -> Read request:  Modes of Operation: Fault
 (000.766851) TX 603 [8]  40 7A 60 00 00 00 00 00 -> Read request:  Target Position -> Why do we care about this?
 (000.773023) RX 583 [8]  43 7A 60 00 00 00 00 00 -> Read responce: Target Position
 (000.874400) TX 603 [8]  2F 00 18 02 01 00 00 00 -> Write request: Transmit PDO: Transmission type: cyclic
 (000.881018) RX 583 [8]  60 00 18 02 00 00 00 00 -> Write response
 (000.982265) TX 603 [8]  2F 01 18 02 01 00 00 00 -> Write request: Transmit PDO: Transmission type: cyclic
 (000.988829) RX 583 [8]  60 01 18 02 00 00 00 00 -> Write response
 (000.990238) TX 000 [2]  01 03                   -> Start Remote Node
 (001.145456) RX 703 [1]  05                      -> Motor heartbeat
 (001.149438) TX 303 [6]  00 01 00 00 00 00       -> PDO: ControlWord: HALT / Target Position: 0.0
 (001.198784) TX 080 [0]                          -> Sync Telegram
 (001.202501) RX 183 [3]  18 02 01                -> PDO: StatusWord: Fault / Modes of Operation Display: Profile Position Mode
 (001.204552) RX 283 [6]  18 02 FF 11 00 00       -> PDO: StatusWord: Fault / Position Actual Value: 4607
 (001.249373) TX 303 [6]  80 01 00 00 00 00       -> PDO: ControlWord: Reset Fault / Target Position: 0.0
 (001.257872) RX 083 [8]  00 00 00 00 00 00 00 00 -> Emergency Telegram: Missing documentation, error fixed?
 (001.298652) TX 080 [0]                          -> Sync Telegram
 (001.302325) RX 183 [3]  50 02 01                -> PDO: StatusWord: Switch On Disabled / Modes of Operation Display: Profile Position Mode
 (001.304471) RX 283 [6]  50 02 FF 11 00 00       -> PDO: StatusWord: Switch On Disabled / Position Actual Value: 4607
 (001.398785) TX 080 [0]                          -> Sync Telegram
 (001.402591) RX 183 [3]  50 02 01                -> PDO: StatusWord: Switch On Disabled / Modes of Operation Display: Profile Position Mode
 (001.404511) RX 283 [6]  50 02 FF 11 00 00       -> PDO: StatusWord: Switch On Disabled / Position Actual Value: 4607
 (001.449247) TX 303 [6]  06 01 00 00 00 00       -> PDO: ControlWord: Shutdown / Target Position: 0.0
 (001.498791) TX 080 [0]                          -> Sync Telegram
 (001.502647) RX 183 [3]  31 02 01                -> PDO: StatusWord: Ready To Switch On / Modes of Operation Display: Profile Position Mode
 (001.504704) RX 283 [6]  31 02 FF 11 00 00       -> PDO: StatusWord: Ready To Switch On / Position Actual Value: 4607
 (001.598786) TX 080 [0]                          -> Sync Telegram
 (001.602503) RX 183 [3]  31 02 01                -> PDO: StatusWord: Ready To Switch On / Modes of Operation Display: Profile Position Mode
 (001.604651) RX 283 [6]  31 02 FE 11 00 00       -> PDO: StatusWord: Ready To Switch On / Position Actual Value: 4606
 (001.649451) TX 303 [6]  07 01 00 00 00 00       -> PDO: ControlWord: Switch On / Target Position: 0.0
 (001.698807) TX 080 [0]                          -> Sync Telegram
 (001.702422) RX 183 [3]  33 02 01                -> PDO: StatusWord: Switched On / Modes of Operation Display: Profile Position Mode
 (001.704698) RX 283 [6]  33 02 FF 11 00 00       -> PDO: StatusWord: Switched On / Position Actual Value: 4607
 (001.798782) TX 080 [0]                          -> Sync Telegram
 (001.802447) RX 183 [3]  33 02 01                -> PDO: StatusWord: Switched On / Modes of Operation Display: Profile Position Mode
[... cut ...]
 (002.609747) RX 283 [6]  37 06 FF 11 00 00       -> PDO: StatusWord: Operation Enabled / Position Actual Value: 4607
etc...
```

As you can see only PDO's are transmitted.

And the output of `rostopic hz` (slightly faster than reported because base_controller was not spawned
```bash
rostopic hz /joint_states 
subscribed to [/joint_states]
average rate: 5.928
	min: 0.157s max: 0.215s std dev: 0.02322s window: 6
average rate: 5.965
	min: 0.157s max: 0.215s std dev: 0.02223s window: 12
average rate: 5.921
	min: 0.157s max: 0.241s std dev: 0.02588s window: 18
```

ipa-mdl, 2017-12-04:
> RX 083 [8]  00 00 00 00 00 00 00 00 -> Emergency Telegram: Missing documentation, no errors?

This means all errors have been resolved.

> The time it takes for the motor to respond to the SDO is 6ms. How can this cause the update rate to be slower than 10hz?

Let's call it a feature ;)
By default the [simple sync master](http://wiki.ros.org/canopen_master#Master_plug-ins) is used, it will wait for a response in the first half and send new commands in the second half of the sync window.
This scheme was less error-prone, but needs fairly fast SDO response times (6ms is really slow!).
In your case the response time is bigger than the step size (5ms) and this leads to synchronisation problems, which delay the updates.

Timple, 2017-12-15:
Still interesting that we cannot obtain an update rate of 10hz because of a 6ms delay in communication. But nevertheless we've managed to configure the PDO's correctly and we now maintain an update rate of 10hz as expected!

For the joint_state_controller that is. 

We do however have troubles initialising the motor using the `canopen_motor_node` because of a different order of initialising (different than performing a manual init using cansend commands). The difference is shown below:

**Manual (working)**
```bash
(5666.297625) TX 000 [2]  81 00                   -> NMT: Reset CAN network
(5666.444813) RX 703 [1]  00                      -> Response from node3
(5666.445997) RX 702 [1]  00                      -> Response from node2
(5666.611252) TX 603 [8]  40 01 10 00 00 00 00 00 -> Read request: Error Register
(5666.613805) RX 583 [8]  4F 01 10 00 00 00 8F AF -> Read response: No error
(5666.634852) TX 603 [8]  2F 00 18 02 01 00 00 00 -> Write request: Set Cyclic transmission PDO1
(5666.637825) RX 583 [8]  60 00 18 02 00 00 00 00 -> Write response
(5666.658574) TX 603 [8]  2F 01 18 02 01 00 00 00 -> Write request: Set Cyclic transmission PDO2
(5666.661475) RX 583 [8]  60 01 18 02 00 00 00 00 -> Write response
(5666.682054) TX 603 [8]  2B 17 10 00 E8 03 00 00 -> Write request: Set heartbeat to 1 sec
(5666.684897) RX 583 [8]  60 17 10 00 00 00 00 00 -> Write response
(5666.704894) TX 603 [8]  2F 01 1A 00 00 00 00 00 -> Write request: Start TPDO remapping, num to 0
(5666.707740) RX 583 [8]  60 01 1A 00 00 00 00 00 -> Write response
(5666.727058) TX 603 [8]  23 01 1A 01 20 00 64 60 -> Write request: TPDO remap position to sub1
(5666.729934) RX 583 [8]  60 01 1A 01 00 00 00 00 -> Write response
(5666.749568) TX 603 [8]  23 01 1A 02 20 00 6C 60 -> Write request: TPDO remap velocity to sub2
(5666.752479) RX 583 [8]  60 01 1A 02 00 00 00 00 -> Write response
(5666.772032) TX 603 [8]  2F 01 1A 00 02 00 00 00 -> Write request: Finish TPDO remapping, num to 2
(5666.775314) RX 583 [8]  60 01 1A 00 00 00 00 00 -> Write response
(5666.794583) TX 603 [8]  2F 01 16 00 00 00 00 00 -> Write request: Start RPDO remapping, num to 0
(5666.797356) RX 583 [8]  60 01 16 00 00 00 00 00 -> Write response
(5666.817243) TX 603 [8]  23 01 16 01 20 00 FF 60 -> Write request: RPDO remap velocity to sub1
(5666.820096) RX 583 [8]  60 01 16 01 00 00 00 00 -> Write response
(5666.839547) TX 603 [8]  2F 01 16 00 01 00 00 00 -> Write request: Finish TPDO remapping, num to 1
(5666.842513) RX 583 [8]  60 01 16 00 00 00 00 00 -> Write response
(5666.861485) TX 603 [8]  2F 60 60 00 03 00 00 00 -> Write request: Set node3 to profile velocity mode
(5666.864235) RX 583 [8]  60 60 60 00 00 00 00 00 -> Write response
(5666.883762) TX 603 [8]  2B 40 60 00 06 00 00 00 -> Write request: Statemachine Shutdown
(5666.886453) RX 583 [8]  60 40 60 00 00 00 00 00 -> Write response
(5666.905863) TX 603 [8]  2B 40 60 00 07 00 00 00 -> Write request: Statemachine Switch on
(5666.908552) RX 583 [8]  60 40 60 00 00 00 00 00 -> Write response
(5666.928583) TX 603 [8]  2B 40 60 00 0F 00 00 00 -> Write request: Statemachine Enable operation
(5666.931290) RX 583 [8]  60 40 60 00 00 00 00 00 -> Write response
(5666.949819) TX 000 [2]  01 03                   -> NMT: Set node3 to mode OPERATIONAL
(5666.973201) TX 303 [8]  FF 00 00 00 00 00 00 00 -> TPDO: Velocity setpoint to FF
(5666.995036) TX 603 [8]  40 FF 60 00 00 00 00 00 -> Read request: Velocity setpoint
(5666.997642) RX 583 [8]  43 FF 60 00 FF 00 00 00 -> Read response: setpoint @ FF
(5667.016074) TX 080 [0]                          -> Send SYNC telegram
(5667.017680) RX 183 [3]  37 06 03                -> RPDO: Statusword: Operation Enabled / Mode of operation: Profile Velocity
(5667.020143) RX 283 [8]  C0 12 00 00 FE 00 00 00 -> RPDO: Position: 4800 / Velocity: 254
(5667.707510) RX 703 [1]  05                      -> Heartbeat: Operational mode
(5668.731459) RX 703 [1]  05                      -> Heartbeat: Operational mode
(5669.755423) RX 703 [1]  05                      -> Heartbeat: Operational mode
(5670.779390) RX 703 [1]  05                      -> Heartbeat: Operational mode
(5671.803379) RX 703 [1]  05                      -> Heartbeat: Operational mode
(5672.032529) TX 603 [8]  23 FF 60 00 00 00 00 00 -> Write request: Velocity setpoint: 0
(5672.035255) RX 583 [8]  60 FF 60 00 00 00 00 00 -> Write Response
(5672.059987) TX 000 [2]  81 00                   -> NMT: Reset CAN network
(5672.207115) RX 703 [1]  00                      -> Response from node3
(5672.208297) RX 702 [1]  00                      -> Response from node2
```

**Using canopen_motor_node**
```bash
(76720.430758) TX 000 [2] 82 03                   -> reset communication node 3
(76720.493416) RX 703 [1] 00                      -> node 3 switched to per-operational
(76720.496462) RX 083 [8] A0 81 01 00 00 00 00 00 -> Emergency Telegram: CAN reset communication
(76720.498918) TX 603 [8] 2B 17 10 00 64 00 00 00 -> Write request: Set heartbeat to 0.1 sec
(76720.502673) RX 583 [8] 60 17 10 00 00 00 00 00 -> Write response
(76720.506572) TX 603 [8] 40 41 60 00 00 00 00 00 -> Read request:  Statusword
(76720.509373) RX 583 [8] 4B 41 60 00 18 02 8F AF -> Read responce: Statusword: Fault
(76720.512892) TX 603 [8] 40 61 60 00 00 00 00 00 -> Read request:  Modes of Operation Display
[... 77 lines cut ...]
(76721.333877) RX 283 [8] 3C 0D 00 00 00 00 00 00 -> TPDO: Actual Position: 3388 / Actual Velocity: 0.0
[... cut ...]
(76721.633891) RX 283 [8] 3C 0D 00 00 00 00 00 00 -> TPDO: Actual Position: 3388 / Actual Velocity: 0.0
(76721.729461) RX 703 [1] 05                      -> Heartbeat Telegram: OPERATIONAL
(76721.730490) TX 080 [0]                         -> Sync telegram
```

The main difference is that we go through the motors internal state machine using SDO instead of PDO (seems logical to me since state switching seems more SDO stuff) and we do this before switching to operational mode.
Is it possible to enforce this behaviour?

ipa-mdl, 2017-12-15:
What's your problem with the second CAN log?
I don't see any errors in it, everyhing seems to be up and running.
The only difference is that you run in position mode.

> The main difference is that we go through the motors internal state machine using SDO instead of PDO (seems logical to me since state switching seems more SDO stuff)

The SDO is just data transfer, The processing of the request might be postponed, so there is not direct relation between successful SDO and successful state switch. (Vendors might implement it differently)

You can use SDOs for the controlword, but since your device processes SDOs rather slowly, it will disturb the sync each time you change the state.

> Still interesting that we cannot obtain an update rate of 10hz because of a 6ms delay in communication.

As I pointed out below the drives expects some response in the half the sync time (so 5ms in your case). The control loop blocks for SDOs and might miss its deadline.

> We do however have troubles initialising the motor using the canopen_motor_node because of a different order of initialising (different than performing a manual init using cansend commands).

Can you elaborate on the different order of initialization? It looks more or less the same to me.
(Your manual order does not follow the standard procedure to disable the PDO before any change).

> and we do this before switching to operational mode.
> Is it possible to enforce this behaviour?

No, at least not that easily. The chain node enforce the initialization of 301 before the initialization of 402. Otherwise you could not use PDOs in the 402 sub routines.

Timple, 2017-12-19:
> What's your problem with the second CAN log?
> I don't see any errors in it, everyhing seems to be up and running.
> The only difference is that you run in position mode.

This is exactly our problem. The motor does not go into velocity mode

> You can use SDOs for the controlword, but since your device processes SDOs rather slowly, it will disturb the sync each time you change the state.

We tried and it indeed did not make a difference.

> (Your manual order does not follow the standard procedure to disable the PDO before any change).

When switching to velocity mode before switching to operation enabled we get the motor in velocity mode. If we switch after operation enabled the motor refuses to switch.
Might this be a wrong implementation of CANopen on the motor side?

ipa-mdl, 2017-12-19:
>When switching to velocity mode before switching to operation enabled we get the motor in velocity mode. 

You might be able to set the ParameterValue in the EDS file for initialization.

Depending on your device you might not be able to change the mode in OPERATION_ENABLED state.
That's why there is the state-switching feature ([example config](https://github.com/ipa320/cob_robots/blob/indigo_dev/cob_hardware_config/robots/cob4-7/config/sensorring_driver.yaml#L10)).

Timple, 2017-12-19:
> You might be able to set the ParameterValue in the EDS file for initialization.

We tried, no succes because it resets after recovering from the communication error (introduced by reset communication)

> Depending on your device you might not be able to change the mode in OPERATION_ENABLED state.
That's why there is the state-switching feature (example config).

We tried that but it does not seem to have any influence. Below you can see the request for velocity mode `2F 60 60 00 03 00 00 00` but the state stays at 5 (OPERATION_ENABLED)

I would expect a request to switch a state back.

```bash 
(1513699392.960446)  can0  TX - -  080   [0]
 (1513699392.963452)  can0  RX - -  183   [3]  37 06 01
 (1513699392.966460)  can0  RX - -  283   [8]  F3 0B 00 00 00 00 00 00
 (1513699393.460455)  can0  TX - -  080   [0]
 (1513699393.463462)  can0  RX - -  183   [3]  37 06 01
 (1513699393.466463)  can0  RX - -  283   [8]  F3 0B 00 00 00 00 00 00
 (1513699393.866477)  can0  RX - -  703   [1]  05
 (1513699393.960463)  can0  TX - -  080   [0]
 (1513699393.963461)  can0  RX - -  183   [3]  37 06 01
 (1513699393.966470)  can0  RX - -  283   [8]  F2 0B 00 00 00 00 00 00
 (1513699394.351459)  can0  TX - -  603   [8]  40 01 10 00 00 00 00 00
 (1513699394.357456)  can0  RX - -  583   [8]  4F 01 10 00 00 00 8F AF
 (1513699394.460458)  can0  TX - -  080   [0]
 (1513699394.463457)  can0  RX - -  183   [3]  37 06 01
 (1513699394.466458)  can0  RX - -  283   [8]  F3 0B 00 00 00 00 00 00
 (1513699394.710461)  can0  TX - -  303   [6]  07 01 00 00 00 00
 (1513699394.890472)  can0  RX - -  703   [1]  05
 (1513699394.960464)  can0  TX - -  080   [0]
 (1513699394.963462)  can0  RX - -  183   [3]  33 02 01
 (1513699394.966464)  can0  RX - -  283   [8]  F3 0B 00 00 00 00 00 00
 (1513699395.210472)  can0  TX - -  603   [8]  2F 60 60 00 03 00 00 00
 (1513699395.216478)  can0  RX - -  583   [8]  60 60 60 00 00 00 00 00
 (1513699395.460470)  can0  TX - -  080   [0]
 (1513699395.463469)  can0  RX - -  183   [3]  33 02 01
 (1513699395.466472)  can0  RX - -  283   [8]  F3 0B 00 00 00 00 00 00
 (1513699395.850479)  can0  TX - -  603   [8]  40 01 10 00 00 00 00 00
 (1513699395.857473)  can0  RX - -  583   [8]  4F 01 10 00 00 00 8F AF
 (1513699395.914485)  can0  RX - -  703   [1]  05
 (1513699395.960480)  can0  TX - -  080   [0]
```

ipa-mdl, 2017-12-19:
> I would expect a request to switch a state back.

It should. But only after the mode actually switched.
Which switching state did you try?
Does the controller's manual include any hint?

> > You might be able to set the ParameterValue in the EDS file for initialization.

> We tried, no succes because it resets to position mode after recovering from the communication error (introduced by reset communication)

Have you tried to modify the source code to do a simple reset?
We had to settle to the communication reset because of problems with Elmo.

Timple, 2018-03-13:
> Have you tried to modify the source code to do a simple reset?

I have to come back at this once we try to integrate the canopen node again. Due to timing we had to resort to a simple hardcoded node to initialise our motors.

> We had to settle to the communication reset because of problems with Elmo.

This is definitely not required for our motors. And since we have more devices on the canbus, it is not even desired behaviour. Could this perhaps be made optional?

ipa-mdl, 2018-03-15:
> We tried, no succes because it resets to position mode after recovering from the communication error (introduced by reset communication)

I don't get why this is an actual issue. All ParameterValue (that are different from DefaultValue) setting will be sent after the reset of communication. So you should be able to set the mode properly, but the vendor migh treat this differently, of course.

>  Could this perhaps be made optional?

I am open to it, but I don't know how.
I guess the best option would be to pass a `Settings` object (filled with ROS parameters) to the Node constructor.
This way we could add a "restart_policy" ("reset_com", "reset" or "skip") parameter.
What do you think?

claudiosouzabrito, 2022-09-29:
> `joint_state_publisher` and the `diff_drive_controller` both run at 2hz while set to 10hz.
> 
> The `joint_state_publisher` starts at about 3-4hz but reduces to 2hz when the `diff_drive_controller` is also enabled.
> 
> This is also the frequency at which the canmotors are operated (I can hear the velocity changes at around 2hz)
> 
> Relevant configs:
> 
> ```yaml
> canopen_motor_chain_node:
>   bus:
>     device: can0
> 
>   sync:
>     interval_ms: 100
>     overflow: 0
> 
>   defaults:
>     eds_file: "config/HBI3760-4/Engel_Drive_1_9_ROS_compatible.eds"
>     dcf_overlay:
>       "1800sub2": "1" # Only send PDO on sync message (Reduces chatter on CANbus)
>       "1801sub2": "1"
>       "1017": "1000" # heartbeat producer [ms]
> 
>   nodes:
>     node2:
>       id: 2
>       name: left_wheel
>       dcf_overlay:
>         "607E": "1" # Reverse Polarity
> 
>     node3:
>       id: 3
>       name: right_wheel
> 
>   diagnostic_period: 1.0
> 
> hardware_interface:
>   joints:
>     - left_wheel
>     - right_wheel
> ```
> 
> ```yaml
> joint_state_controller:
>   type: joint_state_controller/JointStateController
>   publish_rate: 10
> 
> base_controller:
>   type        : "diff_drive_controller/DiffDriveController"
>   open_loop   : false
>   left_wheel  : 'left_wheel'
>   right_wheel : 'right_wheel'
>   publish_rate: 10.0               # default: 50
>   publish_cmd : true
> 
>   wheel_radius        : 0.1
>   wheel_separation    : 0.5
>   base_frame_id       : 'base_link'
>   enable_odom_tf      : false
>   required_drive_mode : 3
> 
> left_wheel_velocity_controller:
>   type: velocity_controllers/JointVelocityController
>   joint: left_wheel
>   required_drive_mode: 3
> 
> right_wheel_velocity_controller:
>   type: velocity_controllers/JointVelocityController
>   joint: right_wheel
>   required_drive_mode: 3
> ```
> 
> * Changing bitrate of the can had no influence
> * Changing the sync interval helped a little bit, could reach 4.7hz
> * Setting PDO to asynchonous with a inhibit time of 20hz did not help
> * The computer does not seem busy (htop <5%) (canbusload <2%)
> * using branch kinetic-devel
> 
> _it seems an issue, if a answers.ros.org is better suited let me know_

Hello, sorry to answer this old message, but can you tell me how you made ROS-cnopen work with diff_controller?