ros-industrial/ros_canopen #265: canopen_motor_node faults

changzhengwu, 2018-01-26:
Thank you sir. 
The process died when launch the node canopen_motor_node, It likely caused by the eds file. When prase the eds file, the process died.
I am new for the canopen eds file configuration. Can you help me. 
Here is the eds file[https://github.com/changzhengwu/eds-file/blob/master/Kollmorgen_RGM_V2.eds](url)
Waiting for your help, thank you.

Timple, 2018-03-13:
Sounds similar to this:
https://github.com/ros-industrial/ros_canopen/issues/256

If you are not familiar with debugging techniques like gdb, perhaps comment parts of your eds file until your node does not segfault ;-)

ipa-mdl, 2018-03-15:
@changzhengwu:

Do you get any message before the ROS node dies? (Try `roslaunch --screen`).
Did you run EDSChk?

@Timple:
The node does not segfault, it just sacrifies itself voluntarily ;)

closed 2020-07-31 by ipa-mdl